ArduPilot/ardupilot #11798: Upgrading AP 3.8rc5 to AP3.9.8 broke aileron (rc5) passthrough in manual mode

marcmerlin, 2019-07-13:
On a Pixhawk1, I had AP 3.8rc5  and was using SERVO5_FUNCTION as aileron in auto modes, and passthrough in manual.
Before the upgrade, I had SERVO5_FUNCTION set to 18, carried over from when it was RC5_FUNCTION.
It worked fine for years, I did have direct RC control in manual mode of channel 5, which I did use for crow flaps (both ailerons going up and flaps going down.
After upgrading to 3.9.8, the parameter was changed from 18 to 4 from what I can tell, but thankfully with a preflight I found out that actuating the crow flaps threw the channel 5 aileron down opposite to channel 1 aileron.  Eventually I figured out that it was not passthrough in manual anymore but avoided an expensive crash by noticing this before taking off.

Now, I can only get back to this working state by going to 3.6.0. With 3.8.0, setting back SERVO5 to 18 makes the servo not work at all.
Setting it to 1 makes it work as passthrough only, but then it doesn't work in FBWA.

I have no idea how I had mode 18 still working up to 3.8rc5, but now it's gone for good unless I get back to 3.6 (3.7 seems removed from the list of available firmwares in MP).
1) is this intentional?
2) can I get back to the working state I used to have
3) is it a bug that this setting died on upgrade without notice and in my case would have a caused a full airframe loss if I hadn't noticed it before takeoff

marcmerlin, 2019-07-13:
@tridge @magicrub I'm still struggling to find out how this was working all the way to 3.8rc5 when I can't get it working in 3.8.0 anymore (can't seem to find a 3.8rc5 build online).
It seems that I did have  SERVO5_FUNCTION set to 18 and it just worked like RC5_FUNCTION 18 used to work before that.
I have many tens of hours spent into getting my setup working just right, and it's been working for years, until it got broken by upgrading to 3.9.8 and now I can't get back to a working state.
I have a flight to do next week where I'd really want to get this back to how it was before so that I don't embark on some brand new flaperon setup all controlled by AP (which was added in the last year, but not have had the time to look at it), as this will take some non trivial time to setup and test.
I'd _much_ rather revert to what I had before and that was known working.
Any idea how I can do that and whether this was really broken between 3.8rc5 and 3.8.0 somehow?

WickedShell, 2019-07-13:
@marcmerlin I suspect you want to investigate the parameter called `MANUAL_RCMASK` I can't quite prove when 3.8RC5 was released but I suspect it was before `MANUAL_RCMASK` was added which is why you are just hitting this now.

The short form is that in 3.8 your input and outputs are decoupled/remapped in all modes including manual. This is all somewhat described here: http://ardupilot.org/plane/docs/plane-3-7-to-3-8-migration.html The parameter above was added as I needed direct control in manual that went past the servo output range.

marcmerlin, 2019-07-13:
Thanks a lot @WickedShell, given that I was running 3.8rc5 and this change happened just afterwards, I never re-read the 3.8 upgrade notes, but only the ones after that and missed MANUAL_RCMASK
Ideally those release notes should have made it very clear that servo mode 18 was removed and would silently get replaced with 4 (which would have caused a crash for me), but that's 2 years ago now, and no one crashed, so I guess it wasn't so bad for everyone else :)
Will test this later this weekend, confirm it works and close this issue. Thank you for the pointer.

marcmerlin, 2019-07-13:
So this was really a corner case, that setting is not really mentioned in the upgrade notes or how mode 18 silently disappeared in version 3.8.
That said, I set servo5 function to 4, it acts as an aileron in FBWA, and then RCMASK to 16, and as soon as I do that, it stops acting as an aileron in FBWA, but then works in MANUAL as expected (with 3.8.0).
It is working in 3.8.5 though.
Then I went to 3.9.8, and it's still working.
Let's close this, I just happened to be using something that was a bit corner case, changed names in a non obvious way, and broken for a little while, but fixed again.
Thanks a mil for the answer because I'm not sure I'd have found it on my own :)

closed 2019-07-13 by marcmerlin