ArduPilot/ardupilot #11802: Provide differential pressure sensor as battery

lthall, 2019-07-14:
I have done some work to look at if a differential pressure sensor can be used as a fuel level sensor. The results suggest that we can get a very nice level reading and correct for accelerations of the airframe too.

The sensor would be set up with a vertical tube in the fuel tank that goes down to the bottom of the tank and a second tube that exits right at the top of the tank to sample the air pressure within the tank. The tubes would be mounted vertically so that fuel does not travel up them normally and large enough so fuel can not be held in them when with surface tension.

To test this I submersed a tube into a water jug and at the end of the log the jug is moved vigorously up and down to simulate a G loading with the flight controller stuck the side of the bottle. The 1cm steps are clearly visible in the raw data suggesting that half cm resolution is easily achievable. The picture below shows the pressure vs the equivalent airspeed to provide a familiar reference.

[image]

You can see that the airspeed jumps immediately from ~12 m/s to zero as we remove the tube from the last cm of water.

Using the vertical acceleration to calculate the depth rather than a simple 9.8m/s/s does a great job of correcting for a banked turn for example. As shown at the end of the log, a simple low pass filter on the pressure and acceleration will also do a good job of removing turbulence from the output.

[image]
[image]
[image]
[image]

auturgy, 2019-07-14:
Somewhat related this branch has a WIP fuel level sensor as a battery, based on  a richen senor
https://github.com/auturgy/ardupilot/tree/FuelLevel_PWM/libraries/AP_BattMonitor

https://richenpower.business.social/product/Level-sensor-water-gasoline-diesel-oil-PWM-output-signal-5V-50Hz-30cmRichenpower

Regards,

James

lthall, 2019-07-14:
I was looking for that sensor as an example. Nice to see you can use a clunk too.

I forgot to add the equation:
Depth = Pressure / (Acceleration * Density)

We could go with an approximate value for density or just leave it as 1 and make the depth relative to full. Making it a 1 full, 0 empty would be consistent with the PWM version above. Doing both would be great!!