ArduPilot/ardupilot #11808: Failure to complete VTOL landing

Hwurzburg, 2019-07-15:
### Bug report

**Issue details**

When executing VTOL landing (as part of mission or as a failsafe ) immediately after a VTOL Loiter, plane can drift away indefinitely without moving from position 1 to position 2 (ie hold altitude indefinitely) in the presence of wind, since pitch is limited to only -3 degs...
https://github.com/ardupilot/ardupilot/blob/master/ArduPlane/quadplane.cpp#L2136

**Version**
Master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane

**Hardware type**
Matek F405- Wing
**Logs**
https://www.dropbox.com/s/scywhb0m543mh8j/NoVTOL_land.bin?dl=0

WickedShell, 2019-07-15:
Why disable the forward motor? This is part of what is expected to help deal with the wind in these scenarios.

Hwurzburg, 2019-07-15:
I don't think this disables the forward motor(or tilt, since this is a tiltrotor)...however, its not on unless Q_VFWD_GAIN is non zero in VTOL modes, as I understand it...and its default is disabled, so its apparently not considered a vital element...in normal VTOL, however, the ability to tilt to the angle limit is usually what is used to hold position or move....but its limited to 3degs forward in this particular scenario however...

WickedShell, 2019-07-15:
The problem with defaulting the VFWD on is it currently needs some tuning in order to safely enable (as well as some scenarios require a rangefinder/terrain to cope well).

There is a comment which explains why the limit is there. I'm sure we are open to improvements on it, but I haven't had a reasonable idea yet for a safe enhancement of what is there.

Hwurzburg, 2019-07-15:
Yeah, Andrew said that he was going to think on it when he discovered it analyzing my Convergence logs....this is easy to reproduce on a Convergence with 6 mph wind nose on...it appears that any VTOL land/RTL/QRTL after moving to a VTOL loiter without more waypoint movements generates the condition of pitch limited to -3deg during landing...if it cannot be fixed, I think a bold warning is needed in the Wiki...

Hwurzburg, 2019-07-20:
Couldn't this pitch limiter be qualified as being active only if groundspeed > Q_WP_SPD?

Hwurzburg, 2019-07-28:
FWIW, this failure also occurs during a normal WP->VTOL Landing if the WP is too far from the landing point and there is wind....it occurred today in 15mph winds when I placed the last waypoint 250feet from landing...no progress against the wind due to pitch being limited...but QLOITER worked file making reasonable progress moving against the wind..

Hwurzburg, 2019-09-11:
resolved with PR #11881 ....closing

closed 2019-09-11 by Hwurzburg