ArduPilot/ardupilot #11814: Sub: SITL will double send some MAV messages

jaylthegreat, 2019-07-15:
### Bug report

**Issue details**

This causes two problems that i've noticed, one is annoying and the other inhibits usefulness.
First is that it will make QGC say some messages twice (for example when you change input gain it says this out loud twice); however, it does not say armed or disarmed twice.

The larger issue this causes is that you cannot upload missions in SITL. You will get some out of sequence error while it's sending them. 
I know that this is caused by the double sending because when I changed the code so that it would ignore every other mission item, the problem was fixed and I could upload missions in SITL.

Note that I could upload missions to my actual sub without issue.

**Version**
dev 3.7.0

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[ X ] Submarine

**Hardware type**
SITL and Pixhawk

WickedShell, 2019-07-15:
How are you using SITL? Are you using the binary directly, or using the sim_vehicle.py script which invokes MAVProxy? The scenario you've described matches experience I've had with MAVProxy forwarding another GCS's telem stream in where it is allowed to try and interact with the mission planning. If you use it in the mode where it doesn't participate, or just use the sitl binary directly I think you will see the problem go away.

jaylthegreat, 2019-07-16:
I've been using sim_vehicle.py.
So you're saying it's a MAVProxy causing the issue?
Regardless, I think the ardusub documentation should mention this or the issue should somehow be solved as it does effect usability.

WickedShell, 2019-07-20:
@jaylthegreat Yeah, that is where I'm pointing the finger. For the record there is a `--no-mavproxy` argument to `sim_vehicle.py` which would also let you use `sim_vehicle.py` without getting the intermediary GCS connected. I've opened a wiki issue to reflect the documentation need https://github.com/ArduPilot/ardupilot_wiki/issues/1851 . If you find that the option I've described here doesn't work then please reopen this and ping me again.

closed 2019-07-20 by WickedShell

jaylthegreat, 2019-07-22:
@WickedShell 
You're correct, it has to do with mavProxy. Using the ```--no-mavproxy``` argument solved my issue.