ArduPilot/ardupilot #11819: Tools: param_parse script does not build files with all params

Hwurzburg, 2019-07-16:
### Bug report

param_parse does not build all parameters into html file

ex: OSDx_VSPEED, _WIND, etc params are not built but others are

**Version**
Master
**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

WickedShell, 2019-07-16:
`param_parse.py` works correctly, The formatting of the comments on `VSPEED` and `WIND` are wrong.

This is wrong:
```
    //@Group: VSPEED    
    //@Path: AP_OSD_Setting.cpp    
    AP_SUBGROUPINFO(vspeed, "VSPEED", 20, AP_OSD_Screen, AP_OSD_Setting),
```

This is correct:
```
    // @Group: VSPEED                                                                                                                                    
    // @Path: AP_OSD_Setting.cpp   
    AP_SUBGROUPINFO(vspeed, "VSPEED", 20, AP_OSD_Screen, AP_OSD_Setting),
```

Note the space between `//` and `@`

khancyr, 2019-07-16:
We should catch this and make it an error in param_parse.py !

Hwurzburg, 2019-08-03:
Since I have resolved this in the PR, shall we close this Pierre, or do you want it left open to entice someone to put in the error check in the script?

DhairyaBahl, 2020-10-05:
I would like to work on this issue. kindly assign it to me.

Thanks

peterbarker, 2020-10-05:
@DhairyaBahl again, can't assign it to you from a technical perspective, but its certainly yours for a while (or until you fix it!) :-)

IamPete1, 2021-08-29:
Not sure of the exact patch that fixed, but I cant replicate on master. It seems like both `//@` and `// @` work the same now.

closed 2021-08-29 by IamPete1