ArduPilot/ardupilot #11841: BRD_IMU_TARGTEMP seems broken in latest master. IMU heating does not stop...

pompecukor, 2019-07-18:
### Bug report

BRD_IMU_TARGTEMP seems to no longer limit heated IMU on Pixhawk2.1

Set it today on 45C and it continued to heat it (in doors air-conditioned room) till I disconnected battery at 69C

Verified by third party, so not my HW/Cube fault. I.E. Tested on other Cube.

_Please describe the problem_

**Version**
Latest master of Arduplane

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Pixhawk Cube

[image]

[image]

BRD_IMU_TARGTEMP=45

tridge, 2019-07-19:
there was a change made to the detection of the polarity of the heater pin by @bugobliterator recently. I have assigned this issue to him

WickedShell, 2019-07-19:
FWIW I can't reproduce this with master as of yesterday. (On plane and a green cube)

pompecukor, 2019-07-20:
> 
> 
> FWIW I can't reproduce this with master as of yesterday. (On plane and a green cube)

Seems very  intermittent, even I find it hard to reproduce on the same Cube. As soon as we removed it from the UAV. However it is real. Have had a couple of confirmations form others, including as you may have seen even Randy experienced this on one of his units.

proficnc, 2019-07-20:
Do you have a log from the original report?
I’m interested to see the temperature on baro 1 vs baro 2
There is a significant self heating on the STM32, the key is to see if the heater was on, or if it’s self heat

pompecukor, 2019-07-20:
> Do you have a log from the original report?

Sorry Philip, I did not have logdisarm enabled. And as mentioned this was office testing. I may have tlog, will have to check Monday on office computer.

pompecukor, 2019-07-20:
I could access the office PC. only tlogs. I did a quick replay. In this case it was set to 45C. Here is from time of startup till over 70C. Roughly a 30min interval.
Was not completely cold from the start, as was already playing around with it before this, but I think you will get the idea. This is chronological obviously.

[image]

[image]

[image]

[image]

[image]

[image]

pompecukor, 2019-07-21:
[image]

pompecukor, 2019-07-21:
Another interesting one. Here I think I was changing the Target temp to see if it help, then rebooting each time. That is where you see the drops in the graph, as the Cube rebooted.

[image]

tridge, 2019-07-27:
@pompecukor can you re-test with the fix from Sid that has now been merged?

pompecukor, 2019-07-27:
Yes I will

IamPete1, 2021-07-26:
Sounds like this has been fixed

closed 2021-07-26 by IamPete1