ArduPilot/ardupilot #11842: External INS for main IMU plus GNSS correction on Pixhawk 2

will-dillingham, 2019-07-18:
Hello! I am new to the drone game but have doing a lot of research over the last few weeks on integrating an external INS as the main source of IMU and GNSS data for the Pixhawk 2. I have the ability to output data from my INS upon power up, sending data is not my problem. I know I will have to input data via the to SPI bus via DF17 connector, but what I cannot find information on is firmware modifications that need to be made to be able to use (ie re-mapping variables) external INS. My motivation for this is to have a much higher accuracy than the on-board IMU provides for the Pixhawk 2 and to use the GNSS receiver to correct data when it can. Perhaps this is only something that can be done using Ardupilot? Like I said, new to the drone game. Thank you!!

IamPete1, 2021-08-15:
We now support external vector nav systems.

closed 2021-08-15 by IamPete1

itsyaboiwillyd, 2021-08-23:
Peter,

Is it sensor integration or just data packets? Can you provide
documentation on this integration? Thank you so much!

Best regards,
*William Dillingham*
Application Engineer
*Inertial **Labs*
39959 Catoctin Ridge Street, Paeonian Springs, VA 20129 U.S.A.
<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fmaps.google.com%2F%3Fq%3D39959%2BCatoctin%2BRidge%2BStreet%2C%2BPaeonian%2BSprings%2C%2BVA%2B20129%2BU.S.A.%26entry%3Dgmail%26source%3Dg&data=01%7C01%7Crafael.carvalho%40wartsila.com%7Ce5f8379a41bd469cecc608d6920721bf%7C37155b7b4b0e488cb3fee84c1d1b7f85%7C0&sdata=rLJ32e4IBiljrPLxs2PicWohitoyiIQvXfB0fKqlLcI%3D&reserved=0>
Mobile: +1-804-625-5709 <%28571%29%20352-3276>
Website: www.inertiallabs.com
<https://emea01.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.inertiallabs.com%2F&data=01%7C01%7Crafael.carvalho%40wartsila.com%7Ce5f8379a41bd469cecc608d6920721bf%7C37155b7b4b0e488cb3fee84c1d1b7f85%7C0&sdata=7NGbgRn%2FqxEYY0tJ3g%2B6YcBeYukTs91uGx%2B4BORdK7g%3D&reserved=0>

rmackay9, 2021-08-23:
@itsyaboiwillyd, I'm afraid we don't have any docs yet for the external ahrs support ([it is on the to-do list though](https://github.com/ArduPilot/ardupilot_wiki/issues/1756)) but in short, AP only supports the vectornav external ahrs and a [couple of parameters are here](https://ardupilot.org/copter/docs/parameters.html#eahrs-parameters).

itsyaboiwillyd, 2023-02-24:
INS-DH-OEM contains our on-board sensors fusion filter, state of the art navigation and guidance algorithms and calibration software.

Maritime 2 (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWC59Z3nJbv5VvQlVW6vf2DWVLsY3L4Xq6MNN880g4y3pds1V1-WJV7CgHPqW1z45S-6bhWlPW3LS5ZD5ZqBP5W1mngGZ2G709mV3zF908LkJgyW2z-kVg7PTNhQN8tF6-HLKytqM44x-Mv13LcW92RMns1hGw0sW5s0mRn28Yp2lW8CwXQH5k73v2W2r-ypV70T6rsW9gb2F36PD2V1W8Jds9M8d9NJqW2Vvt_d3YqqGjW8nVs6P95XCNbW8x_63_6XgwkNW3KBxJz12Lr9RW3d-xVc5B_w4NW4qzYry7V8xbgW2tvXjF3nZsMGN30SJ7dvSPFBW28rBCW8T4KywW5gsxdz3DNfm7W6jv68B8Jw9kPW5kkmQx4CjPNFW15MQnY1zgqDzVBDj_n834YzLW1Cxm9C1MZx7k3myB1 )

Get the best state of the art navigation!

INS-DH-OEM

The Inertial Labs INS-DH-OEM utilizes an advanced dual antenna GNSS receiver, a Honeywell HG4930 IMU which has 3-axes each of calibrated in full operational temperature range Advanced MEMS Accelerometers and new generation of tactical grade MEMS Gyroscopes to provide accurate Position, Velocity, Heading, Pitch and Roll of the device under measure. INS-DH-OEM contains Inertial Labs new on-board sensors fusion filter, state of the art navigation and guidance algorithms and calibration software.

INS-BH-OEM (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWC59Z3nJbv5VvQlVW6vf2DWVLsY3L4Xq6MNN880g3m5mNY1V3Zsc37CgBZ1W59vGnl1x4vB2W10Y2pk8qj-9JW8h6Mb47s1LvnVY7WYf2nhYNCW66TXr05ZgV-xW91w6Dg2G372cW94nT2590PLlFW7X9xVz7pr_8kW2jyZDT1jjfNcN363ccdCGLVSW79Jq5J5qQTxhW2qPB0R4mSWsyW9kGr_M2NS4tkW6wXRcq6WfpDjN2_v_3P8Z60nW2B8Rxb430vp3W1C7dBW3Qm8vfV42b7y6329nKW2_jWJ69dkztBM4sgGYQ4QjdW5HD2wJ4htQQWN6l0dkFK5WcBW4NKXG81cy3kXW9dY2Cd5SbdjgW4MYj9N59VCJ6W2Q8pLN1C5LWwW8wcwld4Ynn8nW5vV65F1jsp0ZW1RyB2k18jYMpVB0N-s1NVjjdW4tZWPB6MfHC2W871Fwp8-0XGrW1J4-4Z2HJkNrW40vQcl38smtpW6q_jYl2wCns_W7jT67q5B1qGr3nM61 )

Learn more
(https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWC59Z3nJbv5VvQlVW6vf2DWVLsY3L4Xq6MNN880g3m5mNY1V3Zsc37CgBZ1W59vGnl1x4vB2W10Y2pk8qj-9JW8h6Mb47s1LvnVY7WYf2nhYNCW66TXr05ZgV-xW91w6Dg2G372cW94nT2590PLlFW7X9xVz7pr_8kW2jyZDT1jjfNcN363ccdCGLVSW79Jq5J5qQTxhW2qPB0R4mSWsyW9kGr_M2NS4tkW6wXRcq6WfpDjN2_v_3P8Z60nW2B8Rxb430vp3W1C7dBW3Qm8vfV42b7y6329nKW2_jWJ69dkztBM4sgGYQ4QjdW5HD2wJ4htQQWN6l0dkFK5WcBW4NKXG81cy3kXW9dY2Cd5SbdjgW4MYj9N59VCJ6W2Q8pLN1C5LWwW8wcwld4Ynn8nW5vV65F1jsp0ZW1RyB2k18jYMpVB0N-s1NVjjdW4tZWPB6MfHC2W871Fwp8-0XGrW1J4-4Z2HJkNrW40vQcl38smtpW6q_jYl2wCns_W7jT67q5B1qGr3nM61 )

KEY FEATURES AND FUNCTIONALITY

• Commercially exportable Dual Antenna GPS-Aided Inertial Navigation System

• Small size & light weight: 85.5 x 67.5 x 52.0 mm size and 280-gram weight

• Honeywell HG4930-CA51 IMU

• Dual Antenna NovAtel GNSS Receiver for Highly Accurate Heading & Position

• GPS, GLONASS, GALILEO, BEIDOU and QZSS constellations

• SBAS, DGPS, RTK and PPP correction signals

• Compatibility with most commercially available LiDARs (Velodyne, RIEGL, FARO etc.)

• High precision trigger for optical cameras

• Up to 600 Hz IMU; 200 Hz INS and up to 100 Hz GNSS data rate

• GNSS measurements and IMU raw data for post processing

• Advanced, extendable, embedded Kalman Filter based sensor fusion algorithms

• State-of-the-art algorithms for different dynamic motions of Vessels, Ships, Helicopters, UAV, UUV, UGV,

AGV, ROV, Gimbals and Land Vehicles

• Full temperature calibration of all sensing elements

Contact Us
(https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWC59Z3nJbv5VvQlVW6vf2DWVLsY3L4Xq6MNN880g4y3pds1V1-WJV7CgDWlW5SkK_99dkn6JW3RQC524dggDDW3H1v4Q4NgwyDW7RRKYv1sJShPW7lGtGs4rzKKTW7GPhKk84sWxdW5WJvnr5Q85kRW6s7xj02f3dSgW80HVsj6NGGzhW99H_4m4RWdYwW10v-KS6pSRkkN6GSS_PwJsPHW1Fz0Cq7sQLjFW8LBsxC6T0vglW37zzCT8Fq7GdW51ysbt3XGZRWVxt6Gw5V27klVqbpF_6hL4XlW7JRDmT8dmQj4N8mS7VLpD0BgW6Xq5cF2n8rnYW1lQRNf68h48qVdTdgy49rn9yW7yRcwy62jVqsW6vtvXJ3nX-DBW95syTD4RBcBqVbcMzj11bKztW5m3XTy8PNDHM36nj1 )

Facebook (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWC59Z3nJbv5VvQlVW6vf2DWVLsY3L4Xq6MNN880g4y3pds1V1-WJV7CgJ1bW2Bm1l429K0BtVM5lvB5CwyrSW4gZDs46-gJgWW3F13Kl7cvmRsW3PQ2nJ3-6Nr2VZlHPN1S_X8WW4R6gQj5wJmhcW4PGMlV3z2YXCN5Hy1GZSrkrxVDy6wW1dNLyWW6YCx176vPZbjW7cbyw07MyzqgN7-F3NtmyX5nW8NqJtL5c1CZRW3xcq5m1m_GZBW49nVP47Ygmc9W1S9kz86GHk3xW5zXCZl1JkzhGW4_vrhJ8sTdKBW33HSD978hcc0W5c8Slm1fMkm4W1hLVXv7N6DryVjk9LX5ptjgnW23Vknz23f2DYW3fvY114vhjfnW2-lFDR7P0F51W2z78zt17LHGpW18cY7B6L4ldC3pln1 )

LinkedIn (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWC59Z3nJbv5VvQlVW6vf2DWVLsY3L4Xq6MNN880g4y3pds1V1-WJV7CgXFTW4CKNlJ5qnDSTW81dHrK50FcKnW1SCbHL7GRfqJW5pr4M25Dcy9yW3nPk3Z22rDM5W4k355g4fjQ2XW5104X03Xs5qrW1DC2hM16G-tyW99YLwb5LXBKlVnpBcC7W8v4vW8Cw0wd9dzqyZW6BLF-88ySYDTW7zm-cr4lvnPjW286MfV4TY1P3W1fz3gd23Rd7XW3TLzp69fZhq4VDfZW61tVPwQW7HdM1z3g9c0YW4WFDb68QYX4kW7h8x2M1CcjmwVj8d3L3glNTCW6Rvq5C7pGNH0W3-5vmM38FFn3Vvf3Gl2Ytd_XW6qxMmm2VzvtjW4hjmp67KR03BW1d5NmJ5x2qHxW2R_RnJ5_Lw7m35wJ1 )

Twitter (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWC59Z3nJbv5VvQlVW6vf2DWVLsY3L4Xq6MNN880g4y3pds1V1-WJV7CgDn-N8J_RD-zTqMhW8zYjvW3CQzhnW4JmyKG4Zk-znW3qqrtD4r_t-SN4n2dzsltPhzV9BS2r64WP3BW3qYy785BcPvVW93F3zN12vpjfW7VLLQt4qCVJtW7TyPRB95dXKtW3srrw-4c5B76W1YFNQy33PVF9N8lWBXT9ztVrW3Ywmsx2KXx10W8ppMl_43tc5yW8yFVBG4SMYbRW25F8vX4R5m7-W8lBsZp90hR5gW7Vc9Q_7x4l2VV4YSbG5l43KgTfbcZ7t3mnZW7hHK-N4lG1wJW9jz5lG3k7VgLMX9gjPFJDKjW5XkR_b9hqVXLW5gCvF27P3D0PN4ZFdnCYX1kLW7tGhf24kC65631Qx1 )

YouTube (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWC59Z3nJbv5VvQlVW6vf2DWVLsY3L4Xq6MNN880g4y3pds1V1-WJV7CgJ1bW2Bm1l429K0BtVM5lvB5CwyrSW4gZDs46-gJgWW3F13Kl7cvmRsW3PQ2nJ3-6Nr2VZlHPN1S_X8WW4R6g [... cut ...]
Manage preferences (https://resources.inertiallabs.com/hs/manage-preferences/unsubscribe?languagePreference=en&d=VnBg137Vnk9sW2p76LK3-1rCSW3_ZXmn4fG_szW3P3RQs43YlrPW3R3CTh3K3L6yW4kH10n3KbsvFW3R0gmB3zjh0lW4hlMkh45RlhdW3P28X24hCPvhW43T2P57T6tR6W17f7hv5r5szsW5BzwpS5r_SYJW4P4G2X6gW_t6V28Wxw760ywPMGKd_P20N68W28MqBT2v5jlSW8MQK2p2Mdds4513TTj1&v=3&_hsenc=p2ANqtz-_noc0hBDdeIeAsSWnq-jeXUcD8CDsvbYhRUutlngs4Kp-7TBB78R0rTEg8JuEz51Iy7fDfH1lNfYYlnplSYy2-LbQDHrWEf8FVue7hIMwHQbmXlGaUbRUUV-c4jhxeF5TXKoFxhrDi2A5GV48g8QDxYc4s7w&_hsmi=247422704 )

itsyaboiwillyd, 2023-02-28:
The Remote Sensing Payload Instrument - RESEPI® was designed as a cost-effective solution for highly accurate Remote Sensing applications.

RESEPI HESAI XT32 Laser-1 (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VXjSDH7v7l6fMCLp1ZpnRxhW3Bmyl94XzRxSN11fKXr3pds1V1-WJV7CgSsMW3RYnLQ1N6H0dW7z1SKH4JCD4hVvVwgp8RqV0NW39PVQM5dtWt5W8nj4GM1p-9vSMj1v27Gf31RN5t7NY9kVh6rW24m7Pq4m3bFXV3xSBW4y465FVQwflV87kMN9W3ybpMv33j49QW6l_dT_8KXv1pW8jnTR85PKDsyW4T2H8684KxpdW7xZ-B-91BqZ-W7VcHsN6dvHCkW1-b5gh7hqLYxW26Nh6L8Qs5FCW6lGK9n2hlfHsW2c0SZ02k-dlYW6Sy2BR58r_N3W92gB0X7CxGLCW7XfG1P6dFftBW3Cj9CJ5WD1twW8lhtlY2Zp9J-W31LjjJ8SWDfWW53qN_b8yg4W0N69rxphKGK8l39ss1 )

Highly Accurate Remote Sensing!

RESEPI® LiDAR

The Remote Sensing Payload Instrument - RESEPI® was designed as a cost-effective solution for highly accurate Remote Sensing applications. RESEPI® utilizes a high-performance Inertial Labs GPS-Aided Inertial Navigation System (INS) with Novatel RTK/PPK single or dual antenna GNSS receiver, integrated with a Linux-based processing platform. The processing platform contains a Wi-Fi interface, an external cellular modem to support RTCM corrections, data logging software, and automated postprocessing.

Product Presentation-14 (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VXjSDH7v7l6fMCLp1ZpnRxhW3Bmyl94XzRxSN11fKVZ5mNXLV3Zsc37CgX_yW6wTknh5_LwJ8W6jHkTJ6g9G2gW21SVrC19rthRW3cR0vM328VJjW8hkrbv7S4BTNW4XVdxP8ypJ-FW8yyYDT55Q2yfW1lBDbW4k2K27W6cSPkZ82ZzN3W3ZKpsl8BbKCvW2JsKzH4JCmnFW4m4bdp95z5gRW6lfdCr472zw2W2v327L6k6Q4TW59_6QS3jFX78W3mWRRW4nY51nW10l1JX1g2l1tW18w6D82n_NgGW3yvHRw4WzrzdW66fYtd1z2KtZW8-R8Gt3n2HFyW5MPxgB5VWtMGW8bYQJf3bXlp9N2h5L1PsXVr0W49CHY32KbCWCV8J6Kp4dW-03W3223QR9dPzNsW4KCm7h43G4dcVF_Qny7xlVJxW8w2WlK7Bv0qSW3gkkMv39wFT1VgwtX38P0kpMW98662C5C1YdfW5q3L9b90k_qv3lDq1 )

Learn more
(https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VXjSDH7v7l6fMCLp1ZpnRxhW3Bmyl94XzRxSN11fKWy5mNYkV3Zsc37CgJ3zVC0JfN1MbsM_W54rlry1xy1y6W6_191s2RWwW7W8XgSm63KFGw6N5GrXVMyglcPW1KNBtR4d1-1pW3qLkh040mxlmW7GgKWK371np5W1QLS124sgbW4W138zbj44gJlRW19H3t11XXTc6W5HRps22QlH1VW3djXMW1xdNwBW6_n4M61kW_b8W9gx1Q424YT4vW42XzZn5DJb5BW1-_zKY3n3jXMMRcnPwPLxhTW3dYjjF2gtKqBVBbNDv5NGl8RN1L8L_9FVRY9W5ml71t4MxTTzW867NJp45hg4PN2SLkc5MRY27W8brKW950fw6tW74D1hm6P-CSRW7NhMdm8TMMDmW23qZYr3r7KGtN70K1lY7x4GjW82BCJj8xXNJbW1bFKkc8_7w3WW30mYtL5rcqn1N4BtnJcbkRFZDbDMbpwJhbW6XDVm98zj__VN3HCnVZ212jnN1SyS9ffFvhqW7XryZn8CSX-c3gXn1 )

Many drone models are compatible with RESEPI®

Contact Us
(https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VXjSDH7v7l6fMCLp1ZpnRxhW3Bmyl94XzRxSN11fKXr3pds1V1-WJV7CgWn0W8CQqH13CQq_kW3lCSGg6pL81vN10yCH-TVxw9W96nkMK85sRw7W6T8rNb6zMw9JW1-CJC44vsBfXW7t9qvy9lXbRSW3cQVYy4zycbpW1mFl8H7hMZLTW63fw7M773B-JW4cRgQk21YBWnN4CD0fR43rfgW5W-pGb59g_TzW5mq-yM1VdmxlW2P7cQ_2NJlz9W9lX0TD2z8z9MW7nqYhf8wzqMHF5gqWJ5rlZtN21G6ym1-cD_W7pfjM17tb4WSVbkrMX6rxxGjN13l243rGsRDW7XLY5f6Zyg1wW3yn5B17Qr8vXVc06rJ52-H9YW7vYVXz2fBRlNV-x0Xp7TgFbHW6xblhZ5FP8Q03pLF1 )

Facebook (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VXjSDH7v7l6fMCLp1ZpnRxhW3Bmyl94XzRxSN11fKXr3pds1V1-WJV7CgFdbVmX6Bn4g5_WlN8PpfyMHpx87N2tmJC8W8cBpW2NZD9L2ntHCBW6L9KVs2r_FW_VdXP7P6nx_dYW96sl_m8nm4QKN3VCS2fFSVyfW2b6GWd2vP8bxW727ssR23lXNSN3VvTJcpplPZVGR4ST5ZksfSVlVk2m7sXdPzW85--NP7q8z1LVlPF3n2dq9dwV-b6YN8fT5SpW1NLmWp3d2TC_W5KggKh7JP-fyW31YW1L4ZYg0XW91fnhD8Z9gbCVtkQRp3Lk8TFW9kScSh2mKcKXW3b8ZdK41kFm_W446d4K1zYJv7W4JJmrm3qDrJlN5r7Msphm4p1W1DP9Vm23KRvjW2HbwJ714189M3hfn1 )

LinkedIn (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VXjSDH7v7l6fMCLp1ZpnRxhW3Bmyl94XzRxSN11fKXr3pds1V1-WJV7CgQ62W6hSLK-2P-LS-W30yHHh6R6J3mW1Rpwtz6rFTFlN6DWSFLcSmwBW70YY1x1fCwLLW4s6b5S2GwJkVW4N8vsS1R-v_7TcwpS8VfylJVZmPjt40FtlRW9dmD6z7nhBwGVbYNGj4FjVnjF6J2mSdL0kWW8CsXNs4tbfQ4W3-ShYN6FVmN6W5ySgjF43PcgpW7wK76X2yB1Z3W24YmYC2BYYyZW5KC_vY5xbbpzW7WbZ-S3C__gjW4D5FT945_TrkW6bgFXY5V7rDpW7gDtXh44MQSTN5KdbCBN8Z8pW8Yj-t02KQC7kW40TDwZ5ll83HM1tLnQ2v2DlN8-w0NfrhHRPW6_-7wT8c-62C31qm1 )

Twitter (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VXjSDH7v7l6fMCLp1ZpnRxhW3Bmyl94XzRxSN11fKXr3pds1V1-WJV7CgY8xW6YZ5Lp7vz0sKN7ltCLtKztHxW27Bgh58Jbj15W8C92N85f2Y67W97TPzD4nb12PW6bqmcd4HYD2NW8slrlS96J--9N5Zhp8D5gNK4N50fjYGRTmL1W5ftKGD7pDmpKN1MWnKb7cvKwW2LB3x-670nvtW27Z-kt95XDxdW6y6gxZ3D9-TdW5ZzRNM939JSKV1zK7K2CvFRmW5JdsDL8491PDVyzCnF7zQ3PVVqqLCr5G35T6W58nRSj3BBL23VnrlBR2QPTRgVR-h581h2hmjW2kjDl_9fQ_B9W6Cwwrj2PNh4TW3KJGh153CNrfW5WQp63724BzPW2-HSSh45ZQM1N7P0DykV3LV13l9Z1 )

YouTube (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VXjSDH7v7l6fMCLp1ZpnRxhW3Bmyl94XzRxSN11fKXr3pds1V1-WJV7CgFdbVmX6Bn4g5_WlN8PpfyMHpx87N2tmJC8W8cBpW2NZD9L2ntHCBW6L9KVs2r_FW_VdXP7P6nx_dYW96sl_m8nm4QKN3VCS2fFSVyfW2b6GWd2vP8bxW727ssR23lXNSN3VvTJcpplPZVGR4ST5ZksfSVlVk2m7sXdPzW85--NP7q8z1LVlPF3n2dq9dwV-b6YN8fT5SpW1NLmWp3d2TC_W5KggKh7JP-fyW31YW1L4ZYg0XW91fnhD8Z9gbCVtkQRp3Lk8TFW9kScSh2mKcKXW3b8ZdK41kFm_W446d4K1zYJv7W4JJmrm3qDrJlN5r7Msphm4p1W1DP9Vm23KRvjW2HbwJ714189M3hfn1 )

Inertial Labs, 39959 Catoctin Ridge Street, Paeonian Springs, Virginia 20129, United States, 703-880-4222

Unsubscribe (https://resources.inertiallabs.com/hs/manage-preferences/unsubscribe-all?languagePreference=en&d=VnBg137Vnk9sW2p76LK3-1rCSW3_ZXmn4fG_szW3P3RQs43YlrPW3R3CTh3K3L6yW4kH10n3KbsvFW3R0gmB3zjh0lW4hlMkh45RlhdW3P28X24hCPvhW43T2P57T6tR6W17f7hv8hgf6yW6lBNjr5t-prYW4NGhH17ZMYfMV28PGC87z4cgMGKd_P20N71W6F3-jy44w_dLVM99rB2crBHW51nJcw1&v=3&_hsenc=p2ANqtz--bANohYvQPhZ66QCypJIdGiq6RryITfL9tEhvKvTOEpw6Bx58M88IRRaAKWVdM-wfGcqPLXRN-w08tcWZNdWHk64LarzGV5V7xmkvMMohIn_80IePoq9yKW1klhMuZavsU [... cut ...]
Manage preferences (https://resources.inertiallabs.com/hs/manage-preferences/unsubscribe?languagePreference=en&d=VnBg137Vnk9sW2p76LK3-1rCSW3_ZXmn4fG_szW3P3RQs43YlrPW3R3CTh3K3L6yW4kH10n3KbsvFW3R0gmB3zjh0lW4hlMkh45RlhdW3P28X24hCPvhW43T2P57T6tR6W17f7hv8hgf6yW6lBNjr5t-prYW4NGhH17ZMYfMV28PGC87z4cgMGKd_P20N71W6F3-jy44w_dLVM99rB2crBHW51nJcw1&v=3&_hsenc=p2ANqtz--bANohYvQPhZ66QCypJIdGiq6RryITfL9tEhvKvTOEpw6Bx58M88IRRaAKWVdM-wfGcqPLXRN-w08tcWZNdWHk64LarzGV5V7xmkvMMohIn_80IePoq9yKW1klhMuZavsUViA9YnFYFC8bGLFTIhHNgfl6ZQ&_hsmi=248109841 )

itsyaboiwillyd, 2023-03-08:
Advanced MEMS sensors based, compact, self-contained strapdown, industrial and tactical grade Inertial Measurement Systems.

IMU 1 (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWxR4v8Hgz4CW6W3kjT1K2SgyW6TYKFS4XVSH2N8tN9Dc3pds1V1-WJV7CgBmkW5kvvDh1RkR32W1Bvgv7880l35W3tV-Rc7RKV6hW3vQ164380r8mW5bz9DB4rRp7hW1yKhB65ZnKdwW600bK63vFbdsW5ZgN2p5Nvs2JN7Z2rnv3Wl86N6dP-KVcPGdFW4XGh9X4Y03rDW3N4mlX2xk5Z7W79LHlZ35b-0yW4ck6zG38Crt8W612swj8PvdbBW4--Vd87b4JknW83QSf11Y6qkqW1wcF255Hrd9sW1PLjSp7HR--dW4thg0b12Z0LmW2rhwVn84js8sW6-NS8973K33ZW87WFPX64wTzXW3B7tFk8kyJB-N28Y3R5yZ7SlW3_L5nN60GmypVpYXHD7hh2fmW5KXCdj68xyBL34Gv1 )

Get the Best Stabilization and Guidance

IMU-P

The Inertial Labs Inertial Measurement Unit (IMU-P) is an Advanced MEMS sensors-based compact, self-contained strapdown, industrial and tactical grade Inertial Measurement Systems and Digital Tilt Sensor that measures linear accelerations, angular rates, Pitch & Roll with three-axis high-grade MEMS accelerometers and three-axis tactical grade MEMS gyroscopes. Angular rates and

accelerations are determined with high accuracy for both motionless and dynamic applications.

IMU-P Tactical A_New_Anodization_2022 (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWxR4v8Hgz4CW6W3kjT1K2SgyW6TYKFS4XVSH2N8tN9C15mNY1V3Zsc37CgQtxW6xpNkC58NwzqW5n6XSF5pxrbsW8rcm7C4bxw5jW1GcYN04glqhmVP0-6c3nTJzVW98Q4KG5cM4H1W93-M5J2nT4sdN62xnJq5_0CbW7rPfG927DvQwW32CZ4p8RJxZmN4bPhvkjk9dYW6dp5Sj48zHTqW7DqRqW86MM-wW23rh7F6trLg3W2JgK3T27K0N6W2tK3K54pmyjPW8Qx56q1qXDDnW8Sgn055cRhR2W25d7jn19GjfWW71jn9t3XxzG4W4KwRT42gK0ssW5MKwfs5T_FrTW4xs0HD7zCcwXW8yhN3p8QPsHpW4yRQgz3fGPyXW7lHsQx7BLr1SW8wY3mJ8mCFcBW4RR0rb5xGPB_W3CNX5j25--_YW91326G5vnYLBW2-8b076YdDdKW2X82Y38RSH0MW42B_n64Ch_7-W5HxcGy5vL543W2hm06g5d2jymVppwcg8d-K3v31d21 )

Learn More
(https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWxR4v8Hgz4CW6W3kjT1K2SgyW6TYKFS4XVSH2N8tN9C15mNY1V3Zsc37CgQtxW6xpNkC58NwzqW5n6XSF5pxrbsW8rcm7C4bxw5jW1GcYN04glqhmVP0-6c3nTJzVW98Q4KG5cM4H1W93-M5J2nT4sdN62xnJq5_0CbW7rPfG927DvQwW32CZ4p8RJxZmN4bPhvkjk9dYW6dp5Sj48zHTqW7DqRqW86MM-wW23rh7F6trLg3W2JgK3T27K0N6W2tK3K54pmyjPW8Qx56q1qXDDnW8Sgn055cRhR2W25d7jn19GjfWW71jn9t3XxzG4W4KwRT42gK0ssW5MKwfs5T_FrTW4xs0HD7zCcwXW8yhN3p8QPsHpW4yRQgz3fGPyXW7lHsQx7BLr1SW8wY3mJ8mCFcBW4RR0rb5xGPB_W3CNX5j25--_YW91326G5vnYLBW2-8b076YdDdKW2X82Y38RSH0MW42B_n64Ch_7-W5HxcGy5vL543W2hm06g5d2jymVppwcg8d-K3v31d21 )

Contact us
(https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWxR4v8Hgz4CW6W3kjT1K2SgyW6TYKFS4XVSH2N8tN9Dc3pds1V1-WJV7CgFT7W6BghQH861wwtW4wRSyP3-RDXkN8F3ynSfgsNqW5GkgC59fcZ0jW3fRMw96XHh0kW1NJZn_5mKtJ9W9jc5P53ph7jrW3NZ67n73TxGJW47dqp313YYcWW1mHvFb72sYP4W6D-xSW4k3C7bW399V2G2sjlFVW4kPb3h2mCfPsW8XQkQN7zvbbWMCR2vY952BWW55dBjf1K7Y5NVd6rlF3jY7fcW5GzsG31T3ks7W8R4bTV6kSV3kW20HNm-3vHlDFW4MnMJ616XzncW7gysql3g7G9rW251_VS54YPJ5W3zQFgw43Q0WVW3xFVcM8mq3ZTW8lGx-s2zczVLW2jw-C87VhYPSW54FKDR6M6h-S3cQq1 )

Facebook (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWxR4v8Hgz4CW6W3kjT1K2SgyW6TYKFS4XVSH2N8tN9Dc3pds1V1-WJV7CgV-BW6dc3N0417ShpW4cKN213DC7vCW3k5qYF4z_LllW8TBkL47lJLHYW5Vvysg5QXW01N7cx_LBl3y1rW89r6G793M9YgN2lsswyC2c6rW4JYwh11Jc8j6W6YR0WH599msYW5bcwM-4fm3ZVVyftVW3lQFYKN5ch4b_Z03YlVT0HLm6FfLX4VtT9HS4f5w5nW48FDYh7rpqfWVcHS6h9jGz2rW13ghzQ2gK3X4W3XPtFp4Qgr08VvdN5s3x2qfvW5w4nD_68Hn-NW5VWjkB37qP9dW6dmXZq7klWl4W4XlV1n5YhY5ZW6CnY9X8nH3psW3nDnHK8Ywmg3W635Hrk1sS9V9W6RVkdY6RbTBX2Sj1 )

LinkedIn (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWxR4v8Hgz4CW6W3kjT1K2SgyW6TYKFS4XVSH2N8tN9Dc3pds1V1-WJV7CgGCYW26xF4R6TlNDTW1bY3J98r9L4kN662CdR-TTvCW791vM55xJ3dsW76v3CQ33X7M2W8jCc1h6vRmkxW8g3gcF1Q7Qb1W7FZQrH78gdK8W4wnHQg9h5JQ6W1Kmm-5609_JKW2ZjNgv1MQVMbW10B4lt3gW0sFW680vC98gx_wvW1fYSZB5H3bRTW9dBVSd2YJ1Q8W41PTCp7KLbCHW5F3rTd4zZv_4W1bN92_4qQfmKW65PBFT5bfqy6W1VNJ-m4H4jgFW6BFw323dLrGmW3ZRNSv3G8gm3W8sgFwj37wz3nW7NZYx23ZLkN6W9fWTP83RNM7DW4hDSWF1PlrnMW64sBvg34TPjpW5HCllV8kbMY_3p_31 )

Twitter (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWxR4v8Hgz4CW6W3kjT1K2SgyW6TYKFS4XVSH2N8tN9Dc3pds1V1-WJV7CgQK9W4t1M6x6Kh4VhN3MMT_8vGl_sW1ctBy97L5yrJW3ySxZx8h4sDqW8bW_ht7Nv70-W6VKKB38jHFCRW4b1zRS5FXcRgW8WT0cj5tB57-W8-99Tj196S6jN8hLbGTLSwnKW4-nQtD6QSf7TW6HMjLh8t9K3GW5dRjb_30n1FdMNsvKGGXGWGW8fbF0z1NbKdjW7Rdg7T325jlXW1PtYjf2B7vnVV4fXM13Qsbf7W7V8YjJ6d0wGQW4hTf7j22JkCsW3tBdfC3L1FyMW8CP2lh8FtXdcW8Kn2Sv2rwgRnVBLxgh30p0W0W1SmMkk8RM21FW293Sj93b2VMSW3c682H8BZ-HZN8s6jSTkLnLb38zj1 )

YouTube (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWxR4v8Hgz4CW6W3kjT1K2SgyW6TYKFS4XVSH2N8tN9Dc3pds1V1-WJV7CgV-BW6dc3N0417ShpW4cKN213DC7vCW3k5qYF4z_LllW8TBkL47lJLHYW5Vvysg5QXW01N7cx_LBl3y1rW89r6G793M9YgN2lsswyC2c6rW4JYwh11Jc8j6W6YR0WH599msYW5bcwM-4fm3ZVVyftVW3lQFYKN5ch4b_Z03YlVT0HLm6FfLX4VtT9HS4f5w5nW48FDYh7rpqfWVcHS6h9jGz2rW13ghzQ2gK3X4W3XPtFp4Qgr08VvdN5s3x2qfvW5w4nD_68Hn-NW5VWjkB37qP9dW6dmXZq7klWl4W4XlV1n5YhY5ZW6CnY9X8nH3psW3nDnHK8Ywmg3W635Hrk1sS9V9W6RVkdY6RbTBX2Sj1 )

Inertial Labs, 39959 Catoctin Ridge Street, Paeonian Springs, Virginia 20129, United States, 703-880-4222

Unsubscribe (https://resources.inertiallabs.com/hs/manage-preferences/unsubscribe-all?languagePreference=en&d=VnBg137Vnk9sW2p76LK3-1rCSW3_ZXmn4fG_szW3P3RQs43YlrPW3R3CTh3K3L6yW4kH10n3KbsvFW3R0gmB3zjh0lW4hlMkh45RlhdW3P28X24hCPvhW43T2P57T6tR6W17f7hv6WggNfW8Bddgv5j5GYRW4LDr8D85WT3VV28XkZ8YxnvHMGKd_P20N6fW5xr_bv3cXqGVW7H0d_J4PHV6_51bFDJ1&v=3&_hsenc=p2ANqtz--IISUa5sBN9xYeIlz3mxMYxmBeLIRcVax3chAt4elEL_30mqd83ig4V3t2ykqaRT0wSyXXa8l3ggeas6bYpdkVeo3DVUb208Lb0pSvVQjoXMbHyvqLvZec0lLA97cSFzgIxOiSrh8GJtozDql80MVklTA-oA&_hsmi=249162074 [... cut ...]
Manage preferences (https://resources.inertiallabs.com/hs/manage-preferences/unsubscribe?languagePreference=en&d=VnBg137Vnk9sW2p76LK3-1rCSW3_ZXmn4fG_szW3P3RQs43YlrPW3R3CTh3K3L6yW4kH10n3KbsvFW3R0gmB3zjh0lW4hlMkh45RlhdW3P28X24hCPvhW43T2P57T6tR6W17f7hv6WggNfW8Bddgv5j5GYRW4LDr8D85WT3VV28XkZ8YxnvHMGKd_P20N6fW5xr_bv3cXqGVW7H0d_J4PHV6_51bFDJ1&v=3&_hsenc=p2ANqtz--IISUa5sBN9xYeIlz3mxMYxmBeLIRcVax3chAt4elEL_30mqd83ig4V3t2ykqaRT0wSyXXa8l3ggeas6bYpdkVeo3DVUb208Lb0pSvVQjoXMbHyvqLvZec0lLA97cSFzgIxOiSrh8GJtozDql80MVklTA-oA&_hsmi=249162074 )

itsyaboiwillyd, 2023-03-09:
Continuous Built-in Test (BIT), configurable communications protocols, electromagnetic interference (EMI) protection, and more.

Air 2 (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWg08Y5ZWGDtW3KjB105-kSLNW76My974XY7QQN1YBnw33pds1V1-WJV7CgJmVN7G2BWBPR_7GW1LdVhl5__tQLW15Znp_3VvQm2W7h8nz_5B90LvN3sSqYX11JMNW3J0fHc2sZ77rW41-Tlx8mYkfHW8XHDkS67k8yRW8WChGv8Lh_CDW5vTqkL7PfwtJVF2YQ65s0wrHW70FNgK52vj-qN5t6KrCsc_-jW42sRrk836C9tW7brYdm94pjGxW2wZ8qp8YLtlYV1ljyG3yqdMsW8ljv7h459tKHV-frZv7Ty1z9W8VJM9M1sxQ2KW7t1xNx6NwrQXW7v4vQ44GgR4ZW4W5tg_7qFRY8W1hs-dl70DPSxW7vH6m-3NKCMCMXmt8NMCZSWW2sVz5p4k4l7BW5c_Jb47jtHyp3kqK1 )

Advanced tactical-grade Inertial Measurement Unit

IMU-FI-200C

The IMU-FI-200C Inertial Labs Inertial Measurement Unit is the latest addition to the Advanced Inertial Measurement Units (IMU) family. Revolutionary due to its compact, self-contained strapdown, advanced tactical-grade Inertial Measurement Unit that measures linear accelerations and angular rates with three-axis tactical-grade, closed-loop Fiber-Optic Gyroscopes (FOG) and three-axis high precision MEMS accelerometers in motionless and high dynamic applications.

IMU-FI-200C (angle 1) (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWg08Y5ZWGDtW3KjB105-kSLNW76My974XY7QQN1YBntV5mNY1V3Zsc37CgK3JW1zL-m975rB32MRQ5ctNVT_5VnBCWh3GzDxlW5CD1K03BfrQlW4w71H35fkr3XW62chbX1F5vs2W1tYQqS71m2RTW5JcZzM3BlGlJW7G8s226LN61XW81ZYYg9k4hc-W2yyWCV1cs0MTW2p7TNl6JGX95W25gDfg85Qk7FW1sG5JC8jxjxNN1Q_s2wfFp9-V3yCSN5qm96bW2ktWct1ScXcQMPp_MMZm1m-W4x6lq28Ss5qpW2nMnl_3rmD94W6HN7wJ2scqtjW94PcdS8Vb3hZW5pY3mm45QtXPW75cCjt9lQRvmW7P2sKk32kb0qW8sTP1-4_rTdSW7ckcHc1dShs2W1bSss98mBJlVW3cWHB92ZcVT8W2ypCHY7-pMfkW6Db_ck5ZwV_pW3CX0cY7-p9tDW2RdfWB9f1D3kVQBcHV4WmBlpW5QxPBy2hVSJTW5dXtvD7lZS1X37tf1 )

Learn More
(https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWg08Y5ZWGDtW3KjB105-kSLNW76My974XY7QQN1YBntV5mNY1V3Zsc37CgK3JW1zL-m975rB32MRQ5ctNVT_5VnBCWh3GzDxlW5CD1K03BfrQlW4w71H35fkr3XW62chbX1F5vs2W1tYQqS71m2RTW5JcZzM3BlGlJW7G8s226LN61XW81ZYYg9k4hc-W2yyWCV1cs0MTW2p7TNl6JGX95W25gDfg85Qk7FW1sG5JC8jxjxNN1Q_s2wfFp9-V3yCSN5qm96bW2ktWct1ScXcQMPp_MMZm1m-W4x6lq28Ss5qpW2nMnl_3rmD94W6HN7wJ2scqtjW94PcdS8Vb3hZW5pY3mm45QtXPW75cCjt9lQRvmW7P2sKk32kb0qW8sTP1-4_rTdSW7ckcHc1dShs2W1bSss98mBJlVW3cWHB92ZcVT8W2ypCHY7-pMfkW6Db_ck5ZwV_pW3CX0cY7-p9tDW2RdfWB9f1D3kVQBcHV4WmBlpW5QxPBy2hVSJTW5dXtvD7lZS1X37tf1 )

Contact us
(https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWg08Y5ZWGDtW3KjB105-kSLNW76My974XY7QQN1YBnw33pds1V1-WJV7CgQglW2KH1Gd7c7ZyhW5BmxP29fnvjMW479zr91MGc6QN74B2GfPCvXsW8bDltH4FwT96W3Y49gC4X36KsW8THj0w3nYfYkW36bbQy5WR_6pVKmyPB6my-GqW6wgsMV8mf5jCW56bTW58llKBlW3zjGSZ994yRjW7ZsmDv5RpDLKW50Cz3r7WYFKcN8cxQbkY4YGcMMgqBknd85_W8_t-B_4nN0YkN82Qr4ZGvsZhW8r7q7V4yNfsGW5m51RZ1TndPjW4bXdjT8-Kj1mW33BvTj2x9qc6W97f3s527y_xWV11-Lq885gbZW6nXkDw7qqcyvW7Rr3CY1QLjNRW5KSB8J93MlPVW7fBf0t7pVMB03n-J1 )

Facebook (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWg08Y5ZWGDtW3KjB105-kSLNW76My974XY7QQN1YBnw33pds1V1-WJV7Cg-HyV-kvZt6R-T0JW8wgy9Z6GhjSVW4Pbv1B7QyzKWW1n73tp5HRh5XW4c6MKJ85GtlDN8d-WS6XcpVrW13kVvv7LzRf3W6lcG2k2lTx7YW1NC7FP2HtMr_N3MPjjp3M2PwT55Wq3-jLQkW61T_NM3d1jykW1d_S2h5bpb80W83NlfP7D7MsVN6F63kt8-rGzW2Fbj4881w45mW5m_rmY2v7sDhN5drzbkQzM--W32Ywmr198RWYVJ1XZR4jLb1YW2ZRqZK2qrdK2W8WjwYy3h391tMg-Xr8tD4XGW8GrthG6Qr9d-W5tjXPw3RR8PLW7T9lmc7gQhvkW7M3-dB4RWS3XW5wgwsy7Wqs983qqt1 )

LinkedIn (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWg08Y5ZWGDtW3KjB105-kSLNW76My974XY7QQN1YBnw33pds1V1-WJV7CgM--W2vsds37hqY58W484j2-1PP4B9V1clfz4Mk5J7W8cvn_46zDjcQW5c6pqn4fwffwW3vpYtn6mLmx1W81zZB01P8Px_W9lBxZT77VH_3N13c9vHyV1ChW8FMPDc7b9pPDW844W375mK18xW7mRjCs68xRT_W5WRFGz83kT_BW3yfkYS4tbrJbW4DtYCd8D7KD0W8ygg6f8MblbHW6DF1N74x6z3LW1MRZCf5sjGFlW4b-7kd3m4Kk3W65GV1T64KNQyN5F2cP7_nr8yW1mvVST8GM_xZW6VzlxL8hHCQrW6RMT3X5mz16cN3QZqKKFYDkcW7J2tqz7xxb7FMQSYy4n1tkbVWN5TX5d6m2-31zF1 )

Twitter (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWg08Y5ZWGDtW3KjB105-kSLNW76My974XY7QQN1YBnw33pds1V1-WJV7CgQ4yW3YLpVj9dTK_0W6wBTNn5Yc9x4VDZh_y8ynvfrW71gFvB2y6mGDW8nKhNY4s5DGRW85jQpB1mgglPW5wS12Q6khPVmW5-Wly43cNSzGW11kLXD6--4n-VfMXg_4HWr0HW2KJNJK2P54PDW2PtGNm3fLbqKW33QtkP7rpNR-W2_3Cdk540M6wW69vKL61Z_4-bW6h3s3F5z1M9zN5My2_1DppXZW2FXz0j67wHfbW3Z44Q92ZvXnzW8pqGws1JDrkSVQ6Drj2dSbJyW6d1p_z8WH02pW7l1Qtp9kWgCpW95sPfs13k973N6--dstst24HW4SYmLd61hCmWW78kfN63ZT2B0W2ZbDh3679mkM34my1 )

YouTube (https://resources.inertiallabs.com/e3t/Ctc/2L+113/cPqFJ04/VWg08Y5ZWGDtW3KjB105-kSLNW76My974XY7QQN1YBnwm3pdskV1-WJV7CgWL3W65MShs9dv4P9Vdb-qn9jZZl2W8pWQWS8rpnvfW5M_XFk1nq-h8W9cyGRr6w8b7-W8CySPv6Ndb7MW4v4qG42_d_w0W5sj7BZ1CxDkMN4dqVy0r4VTgN5GXPWsKbq6gW13YXCq4F1t-sN3sthk-X1RZZW1Ktlsl2fQ-DCW48GhMW4M3vwvW6LknYQ92TzWcW8vB25G3Qb0qrVgwlTc6hC4Q_W57nYvJ3vtnphW1kXgQB7DzV0QVJ4GlT6FfpBcW5z5Wxs86cBB1F2ZJfBVtr7xW2h4NY12fsjyKN1rjLBTgK4jDW2-ggps3y5GbjW2ZJkRj72m145W1HdXJd3DGjRpVZDV2g42Dr4MVwXr9V2hrgm1W2-3bsY1myz6Y3qfT1 )

Inertial Labs, 39959 Catoctin Ridge Street, Paeonian Springs, Virginia 20129, United States, 703-880-4222

Unsubscribe (https://resources.inertiallabs.com/hs/manage-preferences/unsubscribe-all?languagePreference=en&d=VnzSyr7Vnk9sW2p76LK3-1rCSW3_ZXmn4fG_szW3P3RQs43YlrPW3R3CTh3K3L6yW4kH10n3KbsvFW3R0gmB3zjh0lW4hlMkh45RlhdW3P28X24hCPvhW43T2P57T6tR6W15gBJz78tJ9rW6FNtpD4Tr5XmW5wTbV07NnpKSVWMShg6RQXTYMXk_7QbvV2vVYlfjf3tXP35W1WCLsv5SRJ1x3plV2&v=3&_hsenc=p2ANqtz-_jz44ls-Gere_M6wRezhltNVGSwKw3ggrAJcaznnI_9aWgBLREftQeaP77qb3a-zPPutA8Olkrv_eGtRbI7JF9VVqUNSCoPi1IoayXbY-1RwUEXacdCa3jxRif39tCia854iM-7H0Gnqs1wxt_wdtZYIiPmg&_hsmi=249524135 )
[... cut ...]
Manage preferences (https://resources.inertiallabs.com/hs/manage-preferences/unsubscribe?languagePreference=en&d=VnzSyr7Vnk9sW2p76LK3-1rCSW3_ZXmn4fG_szW3P3RQs43YlrPW3R3CTh3K3L6yW4kH10n3KbsvFW3R0gmB3zjh0lW4hlMkh45RlhdW3P28X24hCPvhW43T2P57T6tR6W15gBJz78tJ9rW6FNtpD4Tr5XmW5wTbV07NnpKSVWMShg6RQXTYMXk_7QbvV2vVYlfjf3tXP35W1WCLsv5SRJ1x3plV2&v=3&_hsenc=p2ANqtz-_jz44ls-Gere_M6wRezhltNVGSwKw3ggrAJcaznnI_9aWgBLREftQeaP77qb3a-zPPutA8Olkrv_eGtRbI7JF9VVqUNSCoPi1IoayXbY-1RwUEXacdCa3jxRif39tCia854iM-7H0Gnqs1wxt_wdtZYIiPmg&_hsmi=249524135 )

IamPete1, 2023-03-09:
I'm going to lock this conversation, seems its just getting spammed.