ArduPilot/ardupilot #11850: Irregularly GPS positions in FrSky telemetry output after firmware 3.94 on MatekF450-Wing

Target0815, 2019-07-20:
S.Port (and Passthrough) Telemetry via FrSky RX still works fine with ArduPlane firmware 3.94. Starting with firmware 3.95 incl. the “Latest” firmware GPS positions are only transmitted irregularly. There are pauses of more than 10 seconds and often sensor lost messages.

The failures can be seen on the FrSky transmitter on the telemetry screen (the asterisk does not appear regularly at the GPS position).

https://workupload.com/file/cQV4zZkM (look at 0:47 seconds)

Hardware constellation is Matek F405-Wing. UART via inverter on FrSky-RX (several tested).

The generated S.Port signal on the UART is conspicuous:
FW 3.94: https://workupload.com/file/BDRMyf5u
FW 3.98: https://workupload.com/file/sYUqQQR5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

Target0815, 2019-07-20:
More info:

The GPS works normally (Mission Planner, OSD).

It doesn't matter if the system is armed or disarmed. 

There is no SD card inserted in the F405 wing.

tridge, 2019-09-01:
as the bug was in 3.9.5 I've prepared 3 firmwares between 3.9.4 and 3.9.5 for testing to narrow this down:
http://uav.tridgell.net/Target0815/

yaapu, 2019-09-18:
@Target0815 can you confirm the issue is fixed with https://github.com/ArduPilot/ardupilot/pull/12212

Target0815, 2019-09-19:
@yaapu Yes, I can confirm that the problem with the fix has been solved.

closed 2019-09-19 by Target0815