ArduPilot/ardupilot #11852: H_COL_MID not working on heliquad frame

sokerer, 2019-07-21:
### Bug report

**Issue details**

As can be seen on the attached log, on the heliquad firmware during YAW,  rcoutputs are limited to the lower value of 1500,  most probably the code assumes this is a 0 lift postion, But  should be H_COL_MID which dictates the 0 lift angle (in my parameters H_COL_MID=1270)
[image]

**Version**
3.6.9

**Platform**
[  ] All
[  ] AntennaTracker
[*] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Heliquad

**Hardware type**
Cube black

**Logs**
https://drive.google.com/file/d/17BCRLF7nQ6ggI3oARVHNNWJUswQbCDtV/view

bnsgeyer, 2019-07-21:
@sokerer is correct.   After reviewing the code, it assumes that the 0 deg collective position is at mid servo range.  The code requires knowledge of the 0 deg collective point in order to determine inputs for yaw when operating at low collective settings.  Therefore currently the code requires the user to have equal amounts of positive and negative collective pitch.  The code should use H_MID_COL parameter to determine when collective pitch is 0 deg.

rmackay9, 2020-02-20:
This is fixed in master and will be included in Copter-4.0.3-rc1 which should go out for beta testing today, thanks!

closed 2020-02-20 by rmackay9