ArduPilot/ardupilot #11859: Direct control of drone motor speed through mavlink

JonathanPlasse, 2019-07-22:
### Feature request

I want to be able to control directly the speed  of each individual motor from the companion computer of the drone.

Could it be done by creating a new mavlink message?

**Platform**
- [ ] All
- [ ] AntennaTracker
- [x] Copter
- [ ] Plane
- [ ] Rover
- [ ] Submarine

magicrub, 2019-07-22:
Can you describe the application?

rmackay9, 2019-07-22:
@JonathanPlasse, it's possible using the motor test feature although only for a few seconds..

peterbarker, 2019-07-22:
On Mon, 22 Jul 2019, Randy Mackay wrote:

> @JonathanPlasse, it's possible using the motor test feature although only
> for a few seconds..

I think notionally you could use pass-through using RC channel options.

Note that this is typically a terrible, terrible idea as the latencies 
(and jitter!) will give you terrible control.

rmackay9, 2019-07-22:
The suggestion to use rc passthrough (i.e. set SERVOx_FUNCTION = RCIN1) is OK although this would mean the transmitter couldn't be used because it would be overwritten by the RC_OVERRIDE input from the companion computer.  Also the regular AP flight code couldn't control the vehicle because it would be looking for SERVOx_FUNCTION = motor1,2,3,etc).

JonathanPlasse, 2019-07-23:
> Can you describe the application?

@magicrub, It is to test low level control algorithms as MPC on the companion board.

magicrub, 2019-07-23:
Well then just map an input to an output instead of to a motor. ServoX_function=RCinYpassthrough.

closed 2019-07-23 by magicrub

JonathanPlasse, 2019-07-23:
And to control with the companion computer, can I use `RC_CHANNELS_OVERRIDE`?

magicrub, 2019-07-23:
@JonathanPlasse  yes

JonathanPlasse, 2019-07-23:
Thanks a lot