ArduPilot/ardupilot #11872: Copter: support off-center loading as a parameter

ajshank, 2019-07-23:
### Feature request

For multicopters that are intentionally loaded such that their center of mass is off the geometric center, it would be useful to have a parameter (?) that can specify this [x, y, z] offset. Somewhat akin to a trim value on the RC, this would be a constant bias on the corresponding motor PWMs.
 It seems that right now, the Copter simply "reacts all the time" by compensating for the effective tilt caused by this loading. 

Alternatives:
I wonder if this "bias" can be learnt in-flight, something like:
`if <higher pwm on some axis under level targets>
then "learn biases"`

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

### Possible Solution
I am not 100% sure _where_ exactly this would fit. I have tried fiddling with constant offsets added to motor PWMs directly: ugly and not exactly reliable. If people have ideas, I am happy to look into this further. Also, apologies if I am missing something fundamental here ..

IamPete1, 2019-07-23:
is this not just the Integral of your PID's?

rmackay9, 2019-07-23:
Yes, I think this is just handled by the PID's integrator.  @ajshank I think what we already have is actually better than what you're requesting because the integrator builds up as necessary instead of being a hard-coded user supplied value.  So I think I'll close this, hope that's OK.

closed 2019-07-23 by rmackay9

ajshank, 2019-07-23:
@IamPete1 @rmackay9 Okay, if you guys think so. It seems like that would still need some retuning of the PID values. Not fully convinced that there mightn't be a better solution.
However, thanks for your inputs!

lthall, 2019-07-24:
I can confirm that this is currently handled with the PID settings.
I will also point out that deliberately designing a frame with an offset CG vs the centre of lift breaks the rules for all our motor mixers and will result in thrust coupling with the roll or pitch axis. 

To support these types of trust arrangement we would need to completely rewrite the motor mixers to do it properly. Further they would be dramatically more complex.

So this is generally very bad practice to do on ArduCopter and building a new frame class that supports this would be extremely time consuming if we did it properly.

ajshank, 2019-07-24:
@lthall I take it that, for purely scientific reasons of curiosity, one should look into the mixer code? :)
I am pretty convinced now that for most off-balanced copters, the "I" term will be enough; and even for a couple of my setups, I've found that PID _will_ regulate the attitude. This has needed [auto]retuning of the values so that the rates work well. My observation that some motor(s) would always need to spin a little faster led me to think of learnt biases/offsets as a solution. I see now that this probably needn't be a common feature built into Copter.