ArduPilot/ardupilot #11880: ARSPD_BUS : value of this parameter does not change anything for MS4525DO

StephBMG, 2019-07-25:
The value of ARSPD_BUS parameter does not change anything :

Tested with  mRo I2C Digital Airspeed Sensor MS4525DO

ARSPD_BUS =0, 1 or 2 : the Airspeed Sensor is always detected

Tested on the 3 I2C ports: GPS1, GPS2 and I2C 2 ports, nothing changes, the Airspeed Sensor is always detected 

This also is a problem for dual speed sensors on I2C ports

**Version**
arduplane 3.9.8

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quadplane

**Hardware type**
pixhawk 2.1

WickedShell, 2019-07-25:
Due to how the MS4525 driver was implemented you cannot use the ARSPD_BUS parameter, or multiple versions of the sensor on the same board at the moment. It currently scans for them using a hard coded look up table for the bus and cycles through it in order for each driver. The net effect is it just detects the same sensor twice.

StephBMG, 2019-07-25:
thank you, I'm trying to find the right sensors to get redundant airspeed sensors system, what sensors could work together (UAVCAN, I2C...), can you advise me please ?

tridge, 2019-08-26:
you could use one MS4525 and one MS5525, or one CAN and one MS4525

WickedShell, 2019-08-26:
@tridge This is a real issue. I'm reopening it as it's impossible to use 2 4525's, and you can't use one 4525 and one DLVR unless you know the bus scan order because the 4525 can pick up the DLVR. We really need to come to a path that lets the 4525 work correctly.

msberk, 2024-01-29:
OK, so this has been quiet for 5 years but I just now went through the puzzling but possible means of running two MS4525DOs on two buses on ArduPlane 4.3. Part of me wants to write some docs on this, but also I want to make sure that what's written is what is intended.

My journey in 4 parts:
1) I have one MS4525 plugged into the I2C port of a Pixhawk 6X. It works. **`ARSPD_BUS` shows as 1 in mission planner.**
2) I have two MS4525s plugged into the I2C port of a Pixhawk 6X. "They work" (they do not work due to the I2C address conflict, but Ardupilot reports no errors, and I am not even sure this situation is detectable. The only sign of issues was that there were extremely odd airspeed readings which I imagine came from the two sensors writing the data byte at the same time and I think I2C logical ANDs in this case?). **`ARSPD_BUS` and `ARSPD2_BUS` both show 1.**
3) I realize the two sensors can't be on the same bus, and move the second sensor to the I2C bus on the GPS2 port (I2C 2 according to the port diagram from Holybro). At this point I realize that some sort of auto scanning must be taking place and I try to set the BUS values correctly but make the incorrect assumption due to the lack of a "bus 3" option in `ARSPD(2)_BUS` that the internal bus is excluded and everything is offset by one. I removed the second sensor tried to revert back to step 1 with all parameters the same as that step and now nothing works, at all. Cue a lost work day. 
4) Finally I find [this post](https://github.com/ArduPilot/ardupilot/issues/9715#issuecomment-483450915) which makes me realize that maybe I need to set this parameter out of bounds and suddenly the first sensor works again. Excited, I plug in the second, and set `ARSPD2_BUS` to Bus2, and start playing with them. While I'm not getting message feedback indicating it is working, in practice when I unplug the primary sensor the secondary immediately starts working, and when I replug the primary it takes over again. It seems to work!

This working is all duct-tape, fishing line, and prayers as far as I'm concerned. I see three serious problems:
1) The behavior the MS4525 follows silently changes upon touching the `ARSPD(2)_BUS` parameter, permanently. It goes from autoscanning to respecting it. This seems to be related to a change [here](https://github.com/ArduPilot/ardupilot/commit/e6c7970a19e689f3d005ce31fcb3dc8c77cd4948) from well after the last comment on this open issue. I understand wanting to maintain old behavior so stuff "keeps working" but can we make it explicit? Like `ARSPD_BUS` of -1 is autoscan, and if unconfigured then it gets configured to -1? **This issue matters for both single airspeed and multiple airspeed setups!**
2) `ARSPD(2)_BUS` needs to have more options for `Bus#` added so that it is more clear that one can go to higher values, this got me turned around for a while. **This issue matters for both single airspeed and multiple airspeed setups!**
3) We need some kind of docs written for multiple airspeed sensors. This is where I was starting, and I now have an answer for two MS4525DOs but this seems like it's going to be *very* sensor dependent.

peterbarker, 2026-11-15:
Closed by [PR #17903](https://github.com/ArduPilot/ardupilot/pull/17903). Since [e6c7970a19](https://github.com/ArduPilot/ardupilot/commit/e6c7970a19e689f3d005ce31fcb3dc8c77cd4948) (Plane 4.3) the MS4525 driver probes only the configured bus when ARSPD_BUS/ARSPD2_BUS is set, and also tries addresses 0x36 and 0x46, so two MS4525 sensors on different buses work. The remaining usability point, that an unset BUS still means auto-scan while the GCS shows 1, can be tracked in a new issue if desired.

closed 2026-11-15 by peterbarker