ArduPilot/ardupilot #11888: Copter: Acro Trainer attitude limit only

MrVoltz, 2019-07-26:
### Feature request

**Is your feature request related to a problem? Please describe.**
I would like to be able to use only the angle limiting option of `ACRO_TRAINER`, so it works the same as Acro Trainer in Betaflight. In this mode, no self-leveling would occur and the copter would not lean further than `ANGLE_MAX`.

**Describe the solution you'd like**
I would like to split self leveling rate and angle limiting rate parameters, so I can set one of them to zero. It would bring consistency between Acro mode and Sport mode, too.

In **Acro mode**, around [mode_acro.cpp#L124](https://github.com/ArduPilot/ardupilot/blob/676d75c391926ddc481b5f4d0a6a4efb95145ac8/ArduCopter/mode_acro.cpp#L124) you use `acro_balance_roll` and `acro_balance_pitch` for both angle limiting and self leveling.

On the other hand, in **Sport mode** ([mode_sport.cpp#L57](https://github.com/ArduPilot/ardupilot/blob/676d75c391926ddc481b5f4d0a6a4efb95145ac8/ArduCopter/mode_sport.cpp#L57)), you use `acro_balance_roll` for self-leveling and `acro_rp_p` for angle limiting.

I propose you add two new parameters (for example `acro_anglimit_rate_pitch` and `acro_anglimit_rate_roll`) and use these consistently for both **Acro mode** and **Sport mode**.

**Describe alternatives you've considered**
I've studied the code and there is no way to get my desired behaviour without firmware modification.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2019-07-26:
Thanks for this clear and detailed enhancement request.

lthall, 2019-07-27:
https://github.com/ArduPilot/ardupilot/pull/11895

Can you test this for me?

MrVoltz, 2019-07-30:
Thank you for the quick PR, @lthall ! I've tested #11895 over master in SITL and it is working as expected. Will try in a few days on real Pixracer (I'm learning Acro FPV).

lthall, 2019-08-21:
How did it go?

MrVoltz, 2019-08-21:
It flies without any issues. Thanks again for your patch.

OXINARF, 2019-08-25:
Done by #11895.

closed 2019-08-25 by OXINARF