ArduPilot/ardupilot #11893: Report IMU temp for each IMU via MavLink

JoshWelsh, 2019-07-27:
### Feature request
We currently report RAW_TEMP but this is not directly tied to the IMU.T dataflash message.  This can make monitoring IMU.T difficult from the ground station real time.  Requesting IMU.T (and if appropriate IMU2.T and IMU3.T) be wrapped in to a mavlink message for monitoring through the GCS or companion computer.

**Is your feature request related to a problem? Please describe.**
Yes, the fact that the IMU temp is not explicitly reported to the ground station.

**Describe the solution you'd like**
Wondering if the message `mavlink_raw_imu_t` (and if appropriate, `mavlink_scaled_imu2` and `mavlink_scaled_imu3`) can be extended to include this value?  Would that be the easiest or would that break current GCSs that didn't support the extension?

**Describe alternatives you've considered**
Open to discussion

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
Aware of https://github.com/ArduPilot/ardupilot/issues/11841 but that appears to be a different issue.

proficnc, 2019-07-27:
This would be awesome

WickedShell, 2019-08-13:
Proposed an extension for MAVLink2. https://github.com/mavlink/mavlink/pull/1205

peterbarker, 2020-11-25:
This was done.  New fields in mavlink's IMU messages to convey this.

closed 2020-11-25 by peterbarker