ArduPilot/ardupilot #11894: Log all GUIDED commands (or any navigation commands) as received by GCS/CC to the Dataflash

JoshWelsh, 2019-07-27:
### Feature request

**Is your feature request related to a problem? Please describe.**
There is currently no easy way to verify what command was sent from a GCS or CC with regards to GUIDED coordinates, Altitude changes, speed changes, etc

**Describe the solution you'd like**
I'd like to see a MSG or CMD type entry that logs exactly what the GCS or CC submitted to the autopilot for a navigation/position request.  This would include things like setting a GUIDED waypoint target, a GUIDED waypoint altitude, a change in speed, etc, all as real time commands vs pre-loaded mission items.

As a stretch, I'd also like to see a message sent to the GCS when a CC issues any of these commands, so the GCS can potentially visualize the change in GUIDED target, or speed, or altitude, etc.  This information is more easily derived, however, so isn't as critical during real-time operations.

**Describe alternatives you've considered**
We can review TLOG for `mavlink_position_target_global_int_t` type messages, however this tells us where the autopilot is trying to navigate currently, and we have to try to find when those values change when reviewing logs, and it doesn't show us specifically what command was issued by the GCS or CC, so it leaves us with potential for discrepancy.

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
This really comes down to being able to quickly troubleshoot if a CC app is sending all the correct info for navigation or if there are any issues that can be fixed in a post-flight briefing.  Having a copy of that exact message being sent will assist in these troubleshooting exercises.

JoshWelsh, 2019-11-13:
Hi @tridge we chatted about this a while ago; do you know if anyone has any ideas on how to move forward with this?

peterbarker, 2020-04-11:
We have the `GUID` in a lot of vehicles.  We log mavlink-received commands in `MAVC` and executed mission items in `CMD`.

What's missing?

peterbarker, 2020-11-25:
Pretty sure we've got enough logging here now.

Please reopen if I'm wrong.

closed 2020-11-25 by peterbarker