ArduPilot/ardupilot #11911: Vision Position Bad

maroquio, 2019-07-30:
### Bug report

**Issue details**

I'm using firmware ArduCopter 3.7 downloaded on June 13. My electronic setup is:
Kakute F7 AIO Flight Controller
Benewake TFMini LiDAR
CX-OF or ThoneFlow Optical Flow Sensors (I'm trying both)
Mateksys M8Q-5883 GPS/Compass
FrSky R9 MM Rx Receiver
Everything function I've tested was working perfectly with this firmware, including FlowHold mode with a fantastic position holding. After update to a July version, I started getting "Vision Position Bad" in Mission Planner status tab and the position holding of FlowHold stopped working. Now I flashed back the June 13 firmware and It's working again. So, something was changed in this interval that is causing a "Vision Position Bad" issue which makes FlowHold not working.

**Version**
ArduCopter 3.7 Master (after June 13)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X-quad

**Hardware type**
Kakute F7 AIO + CX-OF/ThoneFlow Optical Flow 

**Logs**
I don't have logs yet.

rmackay9, 2019-07-31:
OK thanks for the report.  Master is constantly changing and hasn't gone through beta testing so issues can creep in.  Hopefully we can get these resolved before entering beta testing of Copter-3.7.

anbello, 2019-07-31:
I am using Copter-3.7 from 8 July on a revo-mini with CX-OF and VL53L1X (as rangefinder).
With EKF3 is all OK, the copter fly in Loiter mode really satble, with EKF2 if I try to switch in Loiter mode I obtain the message:
"Mode change failed: LOITER requires position".

IamPete1, 2021-09-12:
Sorry we never got to this, please re-open if this is still a issue.

closed 2021-09-12 by IamPete1