ArduPilot/ardupilot #11921: RSSI Spektrum SRXL receiver protocol not working

Zadrob, 2019-07-31:
### Bug report

**Issue details**
RSSI for receiver protocol - RSSI_type =3 not working
_Please describe the problem_

I have SPektrum receiver AR7700, with SRXL output. Channels are correctly detected. 
I cant get RSSI reading. It worked on NuttX version (Arduplane 3.9.8) 
**Version**

Arduplane 3.9.8 ChibiOS

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Tiltrotor - Nimbus VTOL

**Hardware type**
Pixhack v3
(RX4 FMU v2)
**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2021-09-12:
@andyp1per another issue in your area.

andyp1per, 2021-09-13:
Outdated protocol rarely used - no interest in fixing

peterbarker, 2024-07-31:
@Zadrob it's been a long time on this issue.

Are you still interested in this at all?

Chances are the problem's gone away - there's been a lot of rewriting!

peterbarker, 2024-09-04:
No answer from @Zadrob and I'd be surprised if this was still a problem in modern firmwares.

Closing thgis, but do re-open if it is still an issue.

closed 2024-09-04 by peterbarker