ArduPilot/ardupilot #11929: Thrust Loss Check: Motor Failure is not detected in Autotune mode

mice5, 2019-08-01:
**Issue details**
After couple minutes of test flight, "I was tuning" the copter in autotune. Roll done correctly, Then after 20 seconds of pitch tuning, copter did strange manouver pitching backward for about 1-2 secs. 
This was the situation, where M3 motor did fail in air - and I didn't notice it, because I couldn't really distinguish it clearly from the tuning procedure and there was no info/event/error from the autopilot. All props were spinning.
Then after next 20 seconds of autotune twitching, second - M4 motor failed and everything fell to the ground. Crash detection disarmed the motors.
Main reason: Hardware issue - ESC overheating.

Software issue: From the code i thing autotune is taken here as some kind of acro/rate mode and it's omitted in the thrust loss check. There was no "Potential Thrust Loss (%d)" message, because the fail wasn't detected by autopilot, and also throttle boost wasn't applied. Motor failure was handled only by Pitch PID.

[image]
_Plot shows RC4=M3, RC5=M4 and Pitch angle._

**Version**
2week old master / ~about Copter-3.6.10 

**Platform**
[X] Copter

**Airframe type**
Y6B

**Hardware type**
Cube / shouldn't matter

**Logs**
Can provide plots imgs if needed.

lthall, 2019-08-01:
Thrust loss detection will only happen if the aircraft does not have enough thrust to maintain altitude with the lost motor. 

The goal of thrust loss detection is to prevent the loss of thrust caused by the increase of the high motor reducing the average thrust from the motors to a lower value than the aircraft needs to maintain height.

This feature is not there to warn the users that one of the motors has failed.

closed 2019-08-01 by lthall

proficnc, 2019-08-01:
I think asking for a message here that reports this situation is quite reasonable. It could save a vehicle. 
Even taking it one step further and triggering an RTL to get it on the ground ASAP. 

Maybe rather than an issue, this should be a feature request?

mice5, 2019-08-01:
My bad. I thought lost motor = thrust loss, so i was expecting the report.
Detecting motor fail would help a lot here. But probably a feature like that would need quite a lot of work.
Should i rewrite the issue to a feature request then?

lthall, 2019-08-01:
> I think asking for a message here that reports this situation is quite reasonable. It could save a vehicle.
> Even taking it one step further and triggering an RTL to get it on the ground ASAP.

I would encourage you to take the time to understand the way the code works and what it can and can not do before telling everybody what is reasonable to do.

As I said, the thrust lost detection relies on a number of checks one of which:
"Thrust loss detection will only happen if the aircraft does not have enough thrust to maintain altitude with the lost motor."

Before answering this and closing the issue I took the time to read through the way I implemented this code and refresh myself on all the various problems I had to avoid. I wanted to see if it was possible to display a warning, even if I could not safely implement thrust boost (In this case it would not have helped). It is not practical. I designed this code to prevent an overloaded aircraft from losing the ability to maintain altitude and smashing into the ground if it had enough thrust to avoid the crash. The code is explicitly designed to minimise the chance of this code causing a crash.

> Should i rewrite the issue to a feature request then?

The problem is we currently don't have any way of detecting the loss of a motor so I don't think it is sensible to clutter the issue list with a request that is currently not possible. Unless it is accompanied with a through explanation of how it could be done, safely and reliably. (I do not believe you can do this but I will be very happy to be wrong)

I am sure it will not be long before we start supporting esc telemetry feedback and that may provide an ESC failure message and I can guarantee that as soon as that is available there will be a message to the ground station warning the pilot.

rmackay9, 2019-08-01:
We have a couple of open issues on the issue list re detecting motor failure:

- [Detect mid-flight motor/ESC failures](https://github.com/ArduPilot/ardupilot/issues/3456)
- [Copter: give altitude priority over yaw.](https://github.com/ArduPilot/ardupilot/issues/3661)
- [Copter: Detect and handle motor loss for improved redundancy](https://github.com/ArduPilot/ardupilot/issues/7525)

I hope to implement some of this in the next 6 months at least with the ToshibaCAN ESCs which can provide feedback.

proficnc, 2019-08-01:
Thanks Randy