ArduPilot/ardupilot #11934: AutoTune for FlowHold Parameters

maroquio, 2019-08-01:
### Feature request

**Is your feature request related to a problem? Please describe.**
The tuning of FlowHold Flight Mode is not so easy to do and the parameters explanations are not so clear.

**Describe the solution you'd like**
It would be awesome to have an autotune for the FlowHold flight mode parameters (FHLD_*). 

**Describe alternatives you've considered**
Using low filter frequency solves partially, but the position does not hold well. It 

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

lthall, 2019-08-01:
I am sorry but with no indication of how to do an autotune on flow hold this is not a practical issue to raise.

closed 2019-08-01 by lthall

maroquio, 2019-08-01:
Okay. I will do some research and maybe I find something useful that can be applied. Sorry, but I thought this space was for feature request only. Thanks and best regards.

tridge, 2019-08-01:
This is really my issue as I did most of the flowhold code. I agree that it is very difficult to tune. That largely results from the poor height estimation that it uses (it tries to estimate height without using a lidar)

maroquio, 2019-08-01:
Yes... compute a transformation (or homography) matrix from optical flow feature tracking is not so precise, mainly when the sensor quality and the environmental conditions don't help. But I got it working acceptably now. It still varies position a little, but it is acceptable. Regards!