ArduPilot/ardupilot #11947: Copter: detect motor failure using ESCs with telemetry and report to pilot

rmackay9, 2019-08-03:
Some ESCs including the ToshibaCAN and DShot ESCs report RPM back to the flight controller.  We should use this feedback to determine if the ESCs are doing what they have been told (i.e. spinning at some RPM close to what we've requested) and if not warn the pilot.

We should perhaps start with a post-arm, pre-takeoff check that the motors are spinning.

lthall, 2019-08-03:
This would also be a good starting point to let me add this detection to the thrust boost capability in the case of a hex or octo.

marcmerlin, 2019-08-03:
How about also detecting it based on throttle vs amp mismatch as explained in https://github.com/ArduPilot/ardupilot/issues/6533

lthall, 2019-08-03:
Unlike a plane we have 8 motors so the throttle vs amp comparison would need to be much more accurately calibrated. I don't think that is practical for a multirotor.