ArduPilot/ardupilot #11969: NTF display smartport on telem one conflict

Gavsv8, 2019-08-05:
### Bug report

I had an osd and led lights on telem 2 with  Mavlink protocol and frsky smartport for R9 on telem 1 . The strange thing is when I enable an NTF display (ssd1306)  connected to the i2c bus it works but the smartport telemetry on telem 1 stops working. I found a workaround to move the led and osd to telem 1 and smartport to telrem 2 and all works including ssd display. There seems to b a link between Mavlink data on i2c bus to telem 1 serial port.

**Version**
Latest and previous version possibly always present

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe **
hex

**Hardware type**
Pixhawk 2.4.8 

**Logs**
Workaround found

IamPete1, 2021-08-29:
@Gavsv8 any update? Is this still a issue on 4.1?

IamPete1, 2021-09-11:
Please re-open if this is still a issue.

closed 2021-09-11 by IamPete1