ArduPilot/ardupilot #11984: SITL: add sim_vehicle argument to switch to auto and arm when possible

magicrub, 2019-08-06:
I know we can make scripts to do this, but it would be nice if there was a built-in script triggered by a sim_vehicle.py CLI argument that will do this:

wait for string "EKF2 IMU1 is using GPS"  (or whatever other ready-to-arm flag is available)
execute "AUTO"
execute "arm throttle"

peterbarker, 2019-08-06:
I think @davidbuzz might have written a mavproxy module recently which would be adaptable to accomplish this.

rmackay9, 2019-08-06:
Perhaps it won't be enough to satisfy the request but it's possible to add a lua script to the arm the vehicle and then the rover's initial mode parameter could be used to get it into auto I think.  I'm sure we will enhance lua scripting to be able to set the mode in the near future.

magicrub, 2019-08-06:
hmm, yeah.. perhaps a LUA script would be all I need. Good idea @rmackay9 !

davidbuzz, 2019-08-06:
yes, i wrote a in-house mavproxy module recently that i called 'sleepyauto' but it's not been authorised by management to be released for public consumption.    for someone familiar with mavproxy though re-creating something like it should not be hard eg:
module load sleep
sleep 25 mode auto
sleep 30 arm throttle
( which would wait 25 seconds after it's loaded and go to auto, and 5 seconds after than arm/fly )

peterbarker, 2020-01-07:
@magicrub Are you satisfied with the solutions offered here?

Really, if you want to do what you suggest - it would take about 5 minutes to write an autotest to do the same as what you describe anyway.