ArduPilot/ardupilot #12005: GUIDED_ENABLE for arduplane

StephBMG, 2019-08-09:
Issue details
GUIDED_ENABLE is available in MP for arducopter flight plan but not in arduplane
I've also tried to use DO_SET_MODE to set guided mode #but it's not working in the Simulation

Version
1.3.68

Platform
Arduplane 3.9.8

Airframe type
Quadplane

Hardware type
Pixhawk cube

**Additional context**
I’m trying to find a way to set the Guided flight mode into a mission for a vtol plane
We use this mode for precision landing in Africa where we deliver health products by drones for NGOs.

So you know an other way to set this Guided mode into a mission ?
I’ve already tried to use the DO_SET_MODE command
I’ve tested it simulation it does not work and I don’t the the value to use in “mode#”

magicrub, 2019-08-09:
Just start sending guided commandeds. It automatically switches modes

closed 2019-08-09 by magicrub

rmackay9, 2019-08-09:
@magicrub,

I suspect that simply switching to Guided mode is what @StephBMG wants but for completeness there is a difference between Guided and Guided-within-Auto as implemented in Copter and Rover.

Guided-within-Auto allows the companion computer to take-over control for a specified period of time or while the vehicle is within a certain distance of the point it was at when the GUIDED_ENABLED command was executed.  Guided-within-Auto is basically safer than just using Guided because once the time limit has expired or the vehicle has flown too far, it returns to regular auto and continues the mission.

Of course, if the companion computer is reliable it could do the switch to Guided (by sending a COMMAND_LONG with a MAV_CMD_SET_MODE within it) and then once it's done it can send another set-mode command to get the vehicle back to Auto.

magicrub, 2019-08-09:
That sounds good, however that's not how it works in plane. There is no way to automatically revert out of Guided. Once a guided point is reached it loiters while staying in guided.

StephBMG, 2019-08-09:
@magicrub , @rmackay9 
Yes you are right @rmackay9 we need to use this command in auto mode, but no need to revert out of Guided:

The precision landing system (LoLas) is formed of a ground module and a drone module. The ground module is mounted onto the landing structure, whereas the drone module is embedded on the drone host.

We put a waypoint command  inside the LoLaS coverage region (20-100m) and the last waypoint is a GUIDED_ENABLE waypoint
Once the drone gets to his final waypoint, the embedded module will detect it and start send commands to the drone which will already be in a GUIDED mode.

This feature will allow us to bring back blood samples from remote aeras in Africa where there is not hospital.

This project has already began in Madagascar

Is there a easy way to use this feature in plane ?
The DO_SET_MODE command could work to set the Guided mode? 

thank you for your help!

rmackay9, 2019-08-10:
@StephBMG, I'm afraid that in plane, the companion computer will need to set a command to change the vehicle into Guided mode.  So it can send a [COMMAND_LONG](https://mavlink.io/en/messages/common.html#COMMAND_LONG) message with the "command" field set to [MAV_CMD_DO_SET_MODE](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_MODE) (i.e. 176) and the "param1" field set to ["15" for Guided](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/mode.h#L29).  To get back to Auto the companion computer will need to send the same command-long but with Param1 set to "10".

By the way it could be helpful to use dronekit but if you've already got something that understands mavlink using the messages described above should work.

StephBMG, 2019-08-10:
@rmackay9 Thank you for your help and the solution you propose
but we can't use the onboard interface (LoLas) as a companion computer since it is limited for the use of landing.

And the command that gives control of the device to LoLas must come from the flight controller (LoLas as no GPS connected, only ultrasonic beacons)
For the next 12 months we'll have more than hundreds of communities to deliver and we have to simplify the way we set the delivery points, adding a companion computer would be not simple to operate...

Too bad that the command DO_SET_MODE in the flight plan does not work, what is it used for?

I have some knowledge in C ++, do you think it would be possible and not too difficult to use part of the code of the copter that include the Guided_enable feature and include it in the code of the plane?

rmackay9, 2019-08-10:
@StephBMG,

I think it would be quite difficult to implement guided-enabled in Plane.  The architecture of Plane is quite different than Copter and Rover.

Adding support for the do-set-mode command is not difficult but it's also not very safe in my opinion.  I can't speak for Plane but in Copter we don't implement it because it is likely that most users would be confused about its purpose and misuse it rather than it actually being useful.

Perhaps the best option is to use a [Lua script](http://ardupilot.org/dev/docs/lua-scripts.html) although we don't currently have a set-mode binding.  Still, it's on the plan to add this I think.

auturgy, 2019-08-10:
setting Q_GUIDED_MODE = 1 might be what is needed here, as I suspect that in this case that the guided phase is in vtol mode?
@tridge might know better

StephBMG, 2019-08-10:
@auturgy yes the guided phase is in Vtol mode
yes setiing Q_GUIDED_MODE = 1 could be helpful
but we have to find a way to set the Guided mode into the Auto mode 
In Guided mode the aircraft will be controlled by the precision landing system 
@tridge is there a way to use Guided mode into Auto mode ?
thank you

rmackay9, 2019-08-10:
I'm not sure if VTOLs have precision landing (someone was asking for it) but if yes then if the companion computer provides LANDING_TARGET messages then they will be consumed during QLAND (or the mission equivalent).  Anyway, maybe i'll let the plane guys take over and I'll step out of this conversation..

auturgy, 2019-08-10:
Precland in quadplane is still an open PR
https://github.com/ArduPilot/ardupilot/pull/11339

Regards,

James