ArduPilot/ardupilot #12010: IN FULL THROTTLE MODE IS NOT GIVEN THAT MUCH OF SPEED IN FLYING TIME

sampathkumarch, 2019-08-10:
** Am gave a full  throttle in FBWA mode or RTL  mode the throttle motor is not rotaing that much speed 

### Bug report

**Issue details**
am using pixhawk cube black . frame is flyingwing . am initially change the parameters  is L1 period is give 19 and airspeed cruise is 20 then after arspd _autocal gave 1, then after takeoff in FBWA mode then after some altitude (60 meters) a changed to auto tune , then after my plane is speed is very low not going forward . am observed my airspeed ratio is 3.876like that .  why its happen speed automatically reduced    when am give the full throttle the the plane is not going that much of speed .

**Version**
arduplane version 3.95

**Platform**
 plane

**Airframe type**
_What type of airframe (flying wing, )_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube)_

**Logs**
_Please provide a link to any relevant logs that show the issue_
https://drive.google.com/open?id=19iP84VAZ7xljuAOoikoNwr0-MzehlwnA

magicrub, 2019-08-10:
Hi @sampathkumarch , you've posted a help question in the bug report area. Please post in the forums at https://discuss.ardupilot.org

closed 2019-08-10 by magicrub

rmackay9, 2019-08-12:
.. and please don't post in all caps.  It's harder to read and generally not good etiquette.

sampathkumarch, 2019-08-13:
So how to solve this problem?

auturgy, 2019-08-13:
Please post your help request with logs to the user forum, discuss.ardupilot.org

sampathkumarch, 2019-08-13:
Thank you for your response. Am already posted but no reply .what can i do?

auturgy, 2019-08-13:
I’d link your post to this issue, and read through the documentation to find parameters that might limit your speed, max throttle or pwm range. Google might help you find out if others have had similar issues and have resolved them.

Regards,

James

sampathkumarch, 2019-08-13:
Today I reset the  parameter list ,then after i change the  PID values ,then after   I flew in FBWA mode at full throttle it's go 25 to 30m/s . Then after I gave RTL  .it  came to RTL position and circuling but speed gradually decreasing. That is the problem .when am given RTL it's circuling very low speed how can solve this problem please help me

auturgy, 2019-08-13:
Use the forum.