ArduPilot/ardupilot #12012: disarm mavlink command ignored during takeoff

dwiel, 2019-08-10:
### Bug report

**Issue details**

I wanted to abort a failing takeoff from GUIDED mode, and so sent multiple disarm events via MAVLINK. However the drone continued to fly for quite some time afterwards. I believe it only finally gave up after I killed the SLAM script which was providing positioning and so a lack of positioning failsafe kicked into abort, though I'm not 100% sure.

**Version**

master branch as of a few days ago, commit `b2d9d7b6a07ef4bcec0362e8bf404b2ad9f4a489`.

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
HEX

**Hardware type**
mRobotics x2.1

**Logs**
None available at the moment. I can get them in a couple days if it would be helpful.

peterbarker, 2019-08-11:
On Sat, 10 Aug 2019, Zach Dwiel wrote:

> I wanted to abort a failing takeoff from GUIDED mode, and so sent
> multiple disarm events via MAVLINK. However the drone continued to fly
> for quite some time afterwards. I believe it only finally [...]

We deny disarming via mavlink if believe the vehicle is flying.

That being said, we do allow you to force the issue by setting a magic 
value in the relevant mavlink packet - issuing "disarm force" in MAVProxy 
can be useful for creating a tiny meteor, for example.

Yours,
-- 
Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: in what circumstances would you want reverse thrust for takeoff? --tridge

dwiel, 2019-08-12:
perfect, thanks!

closed 2019-08-12 by dwiel