ArduPilot/ardupilot #12013: Implementation of Free Norwegian Elevation data

Supermagnum, 2019-08-10:
Hi, I am curious if it’s possible to implement the free and open elevation DTM10 data from the Norwegian Kartverket aka Norwegian Mapping Authority.
 It’s in 10x10 meter grid measurements, with a accuracy of ± 4 to 6 meters,as it’s superior to any SRTM data.
The DTM50 format is in 50x50 meter grids,same resolution. I don't know if it's possible to downgrade the size of the grids, 50x50 are the largest.

Preprocessed national dataset exports are available.
All datasets are in EUREF89, NN2000. UTM zone 33 has full coverage. Zone 32 and 35 where applicable.
The DTM10 file covering the entire Norway is  18139 MB updated 20.07.2019

DTM50  is 559 MB

The files can be found at:
https://hoydedata.no/LaserInnsyn/#downloadTab

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[x ] Plane
[x  ] Rover
[  ] Submarine

rmackay9, 2019-08-12:
The terrain data is provided by the ground stations rather than being held persistently withing the flight code so my immediate reaction is that this enhancement would need to be done in one of the ground stations (Mission Planner, QGC, MAVProxy, etc)

rmackay9, 2019-08-13:
I think I'll close this if that's OK because I don't think we have any plans to add direct inclusion of terrain data in the flight code, instead the plan is to stick with the GCSs providing this

closed 2019-08-13 by rmackay9