ArduPilot/ardupilot #12014: AP_RANGEFINDER: Rangefinder params need a subgroup rename

Hwurzburg, 2019-08-11:
### Bug report
Rangefinder params are recursed in two sub-groups with the same name, leading to param expansions in the wiki and MP with identical names and wiki build errors...not only ugly but potentially confusing..

**Version**
master

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Hwurzburg, 2019-08-11:
I will happily do the work in the code to change this to eliminate wiki build errors, but need a decision on how the two subgroups should be named...

rmackay9, 2019-08-12:
Here is what it looks like on the wiki.  So we can see there are two expanders for each range finder instance.
[image]

peterbarker, 2020-06-01:
The linked PR just eliminated the duplicate heading here.

closed 2020-06-21 by Hwurzburg