ArduPilot/ardupilot #12025: Copter/Rover: handle SmartRTL failure to get semaphore to avoid Land/Hold

rmackay9, 2019-08-13:
There have been [two reports of SmartRTL (on Copter) triggering a Land before reaching home](https://discuss.ardupilot.org/t/smart-rtl-failure-cause/45928).  In the logs, the SRTL message contained "Action" "5" which is "SRTL_POP_FAILED_NO_SEMAPHORE" which appears [here in the AP_SmartRTL library's pop_point() method](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_SmartRTL/AP_SmartRTL.cpp#L146).

We should do the following:

- investigate why pop_point would be unable to get the semaphore because we expect that no other threads are still trying to save points because the AP_SmartRTL::request_thorough_cleanup() should have only returned true if the background thread was completely done updating the path.
- if we cannot find a bug in the above and are happy to fix the symptom (i.e. the failed semaphore get) we could enhance the ModeSmartRTL::update() function so that it decides it should switch to the SmartRTL_StopAtHome state not based on whether the pop_point fails but rather based on the number of remaining points in the path (i.e. use AP_SmartRTL::get_num_points()).

As a side note, since AP_SmartRTL was written we have moved to using the WITH_SEMAPHORE macro instead of using the semaphore's take_nonblocking() and give() methods.

rmackay9, 2019-08-13:
It's possible that the commits in this branch may fix the issue: https://github.com/rmackay9/rmackay9-ardupilot/tree/smartrtl-semaphore-fix

lukezhqin, 2019-09-30:
how about apply the waypoint nav-controller directly bu reverse the sequence of the waypoint in smart-rtl mode?

lukezhqin, 2019-09-30:
Very interested in this issue here!

rmackay9, 2020-12-18:
I think we've resolved this issue now so I'm going to close it.  FYI @peterbarker

closed 2020-12-18 by rmackay9