ArduPilot/ardupilot #12028: Mount: retract automatically when defined altitude is reached

gediminasgu, 2019-08-13:
### Feature request

**Is your feature request related to a problem? Please describe.**
I saw many times when operators forget to retract the camera mount before landing and usually breaks the camera or mount.

**Describe the solution you'd like**
I'd like to have a parameter where I would set at which altitude (below the defined point) I'd like the mount to be retracted automatically.

**Describe alternatives you've considered**
I didn't find any other alternatives for it how to automate it.

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

lvale, 2019-08-21:
> Describe alternatives you've considered
> I didn't find any other alternatives for it how to automate it.

Quite easy to do on a RC radio that has telemetry like the Taranis. 

Read the height and vertical speed from the telemetry. The time it takes to deploy the LGR is taken into consideration to deploy the LGR.

gediminasgu, 2019-08-21:
Yes, but then there are too many requirements to make it work:
- You need Taranis with Telemetry
- You need to know how to configure it
- You need to have a link with the drone
- There should be enough time for Taranis to react
- You need an additional RC channel to make it work

proficnc, 2019-08-21:
This would be an awesome feature, it’s an expensive mistake!