ArduPilot/ardupilot #12035: Plane: TakeOff can be triggered whilst in forward flight

votingprawn, 2019-08-13:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**

If a "TakeOff" command is sent during active flight, the aircraft will attempt to takeoff again to a new altitude =  (current altitude) + (takeoff altitude).  This is particularly dramatic in a quadplane where it'll spool up the lift motors and try to hover up to that new altitude.

After you have made this mistake once, ArduPlane will throw a "Already flying - no takeoff" error if you try and repeat it.

There is an associated UX issue in MissionPlanner where the takeoff set altitude modal is identical to the "Fly to Here Alt" modal, meaning you can accidentally select TakeOff in the MissionPlanner right click context menu, but think you are just setting a guided mode altitude.  More concerning is that if you press "cancel" on this modal, the aircraft will still try to TakeOff to the default altitude in the modal (i.e. the last guided mode altitude you set).  

To emphasise, if you accidentally click TakeOff in the context menu, there is no way that I have found to escape the following modal that doesn't cause a TakeOff instruction to be sent, leading to a potentially catastrophic outcome.

**Version**
3.9.8 in real life
3.10.0-dev in SITL

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane (conventional FW layout)

**Hardware type**
Pixhawk 2.1 (Cube)

**Logs**
I can provide logs later if required (I need to trim them, and change my trousers after this flight), but it is easily repeatable using the quadplane SITL model in MissionPlanner.

magicrub, 2019-08-13:
Normal (non-vtol) Plane Takeoff did not historically have this behavior, it was always relative to home alt so resetting the mission to wp1 /w a takeoff was OK. So, if this is indeed a current+missionAlt then it's a regression bug. However, I don't have much experience in testing that for vtol takeoff.

Can you confirm this is a vtol-only thing or not?

votingprawn, 2019-08-13:
Just checked in SITL for standard plane and you are correct - the TakeOff command does 'nothing' if the TakeOff Alt is the same as the current altitude, and results in a climb if TakeOff Altitude is greater than current altitude, and a descent if it is lower.

I assume that this is the expected behavior - allowing for missions to be be rerun without needing logic to skip over takeoff commands?

magicrub, 2019-08-13:
Correct. Can you please change the info in your original post to reflect this being vtol only.

votingprawn, 2019-08-13:
Updated - sorry for the inaccuracy.

WickedShell, 2019-08-13:
This is a behavior I disagreed strongly with @tridge but was unable to convince him to change the default. It can be turned off by setting the bit in [Q_OPTIONS](http://ardupilot.org/plane/docs/parameters.html#q-options-quadplane-options) (you want to set the bit for "Respect takeoff frame types") With that set you shouldn't see a problem anymore. (If you are below the takeoff altitude the plane will start a takeoff to the requested altitude, if you are above it then it will just move on to the next waypoint immediately). As a note this is not in the plane stable release, only in master, and this will look at the frame you planned the takeoff in (IE if you plan absolute altitude missions then the takeoff altitude should be above sea level, not home)

I'm going to close this now, but if you have a scenario that doesn't work correctly with that bit set then reopen this or comment on it and we can carry forward.

closed 2019-08-13 by WickedShell