ArduPilot/ardupilot #12036: Rover: Stop exactly on waypoint

magicrub, 2019-08-13:
Stop on a waypoint, exactly. no coasting, no kinda brake or kinda slow down.. Navigation controller needs to accurately aim to stop exactly on a waypoint on command when there's any delay either on the waypoint, or perhaps, the next waypoint is NAV_DELAY.

mikerob, 2019-08-13:
I will be interested to see how wyou tackle this.

Does it motivate a change to using a longitudinal acceleration controller rather than velocity controller?

rmackay9, 2020-07-15:
By the way it may help to add a delay into the mission's waypoint command.  There's an outstanding issue here for the MP's command list to be corrected to show the Delay is the first column.  @meee1!