ArduPilot/ardupilot #12041: SITL Cygwin thread creation fails

IamPete1, 2019-08-14:
In the process of developing path planning stuff I have found that I cannot run @rmackay9 's OA stuff in master. This line https://github.com/ArduPilot/ardupilot/blob/466681a94eed9fee6150e6dddbe170b74bc747e4/libraries/AC_Avoidance/AP_OAPathPlanner.cpp#L138

returns in a 'Failed to set stack of size 16384 for thread avoidance'  possibly there are other thread creation instances where this would also fail. 

Not sure if this is my setup or a wider Cygwin issue.

Thanks

magicrub, 2019-08-14:
It's a wider issue. Building on arm targets like the Nvidia TX2 has the same problem.

magicrub, 2019-08-14:
note, the creation of that thread is a little different than other threads so perhaps it should just be changed to follow the other methods like SRTL

rmackay9, 2019-08-14:
yes, i also see the problem when running on Cygwin under Windows.  It's fine under Ubuntu which is where i do most of my SITL testing but it would be nice to have it working under Cygwin/Windows as well.

magicrub, 2019-08-15:
Randy, can you change the thread creation to be the same as safe rtl and see if that fixes it?

IamPete1, 2019-08-15:
I will test if Smart RTL works under Cygwin, it looks as though the SRTL registers a callback with a existing thread rather than creating a new one. there are a couple of places that use the same method as OA but they are for UAVCAN and may never be hit in SITL.

IamPete1, 2019-08-15:
Smart RTL works as it should

rmackay9, 2019-08-17:
OK, I think we want to creat a new thread for OA though so even if it works to use the SmartRTL method that's not what we want going into master.

IamPete1, 2019-08-21:
closing as this as the issue is now fixed for Cygwin

closed 2019-08-21 by IamPete1