ArduPilot/ardupilot #12044: QGC can not detecting the pixhawk board , B/E in red color?

hughhugh, 2019-08-14:
Hi. I met a problem after uploading the firmware built from the master branch. The codes can be compiled and uploaded. But after it is uploaded, the B/E light is always in orange color. The qgroundcontrol can not detect the pixhawk board now. Has anyone met this problem before?

peterbarker, 2019-08-14:
On Wed, 14 Aug 2019, Nailong Wu wrote:

> Hi. I met a problem after uploading the firmware built from the master branch. The codes can be compiled and uploaded. But after it is uploaded, the B/E
> light is always in orange color. The qgroundcontrol can not detect the pixhawk board now. Has anyone met this problem before?

Which board?

Which git hash?

hughhugh, 2019-08-15:
@peterbarker 
Hi.
I remove the arm-none-eabi-gcc and download the <gcc-arm-none-eabi-6-2017-q2-update-linux.tar.bz2>, set the compiler again manually rather than using <Tools/environment_install/install-prereqs-ubuntu.sh -y>.  
Right now, the QGC can connect the pixhawk board. It works now. I think there is something wrong with my system update which caused the error.

closed 2021-08-15 by IamPete1