ArduPilot/ardupilot #12064: Req: add custom failsafe actions

CodingGhost, 2019-08-17:
#### Issue details
ArduPilot should have a way to set custom failsafe actions, at least failsafe servo/pwm positions for outputs.
an example usecase would be: I have a beeper connected to a pwm output, and want it to beep at a failsafe event. Currently there is no way that I know to do this. with a custom failsafe action, one could set the PWM value for the beeper to the desired values.

#### Version
all

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Flying wing

#### Hardware type
Matek F405-Wing

#### Logs
..-..

khancyr, 2019-08-17:
Hello,

Thanks for the idea. That is interesting. It shouldn't be hard to implement but hard to use because of current failsafe handling.

krzysztofkuczek, 2019-08-17:
Great idea,
it should also handle situation where RC signal is lost (e.g. sbus failsafe flag is set). There should be possible at least to keep rc channel value, or overwrite by defined value.

rmackay9, 2019-08-19:
I think we should probably handling this in Lua scripts.  So this would be a new failsafe action that triggers a script.  The script would likely need to be in a specific filename to work.  I might change the title to make this more clear

Mogway28, 2020-11-03:
Similar problem https://github.com/ArduPilot/ardupilot/issues/13362

joneseym, 2021-01-26:
Here's specific use case for this.  Small long-range FPV quads can have trouble getting enough separation between high power VTx and the GPS module.  This can make the GPS reception poor when the VTx is powered.  If the VTx is switched by a relay output it would be very nice to have the VTx automatically switched off on RC failsafe to provide the best chance of a successful RTL.

CodingGhost, 2021-01-26:
thats a brilliant Idea!

CodingGhost, 2022-05-01:
is there any update on this?

rmackay9, 2022-05-01:
@CodingGhost,

The only update is that I've recently added an EKF failsafe override for Copter implemented in Lua.  It's not a full flexible solution though.

Script: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/copter-deadreckon-home.lua
Test flight: https://www.youtube.com/watch?v=KKShYheW4J0
SITL testing: https://www.youtube.com/watch?v=esM0EqMH_BE