ArduPilot/ardupilot #12078: NO RC Receiver  (but flight controller is getting input from the receiver)

reluctantgithubuser, 2019-08-19:
Forum link:
https://discuss.ardupilot.org/t/no-rc-receiver-lockup-between-flight-controller-and-receiver/45811

### Bug report
Bottom line:

Flight controller in Arduplane reports "No RC Receiver" but flight controller logs indicate it is receiving values from the RC Receiver, including inputs on the throttle channel well above the failsafe setting. Independent logs on the Taranis controller show the RC Transmitter and RC Receiver are in good communication.

A failsafe/no RC receiver should not be reported by the software if it is getting data from the RC receiver. This seems to me a bug.

**Issue details**
Airframe: Nano Goblin 600 mm flying wing

System:
Omnibus F4 Pro V3
Arduplane v 3.9.4 (Chibios)

Companion computer on board (Raspberry Pi) with 4G modem.

Taranis TX and TBS Crossfire RF module (full)
TBS Nano Receiver
Firmware v 2.88 on the TBS Crossfire module
OpenTX firmware 2.2.3 on the Taranis.

Receiver is set to put out 982 on ch3 in event off lost link to transmitter.

Failsafe set for throttle at 990 PWM.

Lowest value of ch3 is 998 PWM when transmitter is connected.

Testing on bench by turning off receiver:
RSSI (channel 8) goes to zero in Mission Planner.
ch3in goes to 982 in Mission Planner.
Throttle failsafe kicks in, as it should.

However, in the air the following happened:
80 seconds into the flight, the Mission Planner called out
NO RC Receiver
I managed to put plane in auto mode through Mission Planner/companion computer connection and it completed the mission and auto- landed safely.
Exactly on touchdown at landing, the failsafe was removed, suggesting maybe something mechanical.
However post flight inspection shows wiring seems to be good on superficial tugging and wiggling and visual inspection.

bin logs and tlogs (through the companion computer) both show about 1500 from rcin ch3 at the time of failsafe event.
bin logs from on board SD card on flight controller and tlog file from Mission Planner available at:
https://discuss.ardupilot.org/uploads/default/original/3X/e/2/e2de4f96445c41bb39de162da0afc92f96bc265d.BIN

graph at:
https://drive.google.com/file/d/1a-cVj-KC6sMdFLv5ItK2pvBKNJ3eWsBm/view?usp=drive_open

tlog:
https://drive.google.com/file/d/1V2jvoW2MNSqxyONqdYXMmztqNOC4cA4n/view?usp=drive_open

with these graphs:
https://discuss.ardupilot.org/uploads/default/original/3X/f/0/f0796ad5c4eaa3b05aa72f3d61448af9a37bf0fe.png

https://discuss.ardupilot.org/uploads/default/original/3X/3/6/369477d45bdb0bef0cf914548cdb9521fc2cbd75.png

https://discuss.ardupilot.org/uploads/default/original/3X/c/2/c21ab1a5f3e9d13b532442e306d8258f22756e78.png

THE PROBLEM as I see it is that ch3 raw is giving 1500 or so out, and ch8 (rssi channel) is also giving a high value. Therefore, why did the flight controller Ardupilot software call out a failsafe/NO RC Receiver/Throttle failsafe even?

It is even more confusing because as I show below, the receiver was never out of contact with the Taranis radio as verified by separate telemetry log in the SD card in the Taranis radio which records the RSSI at the radio receiver in the plane, and it was good.

Specifically: logs of the Taranis show the RSSI at the receiver was fine and data was being received by the receiver. (TBS RF module in the radio on the ground has some telemetry with the on-plane radio receiver). Therefore, the radio reciever on the plane was in good contact with the radio on the ground.
image of RSSI etc here:

https://discuss.ardupilot.org/uploads/default/original/3X/7/5/75aee83abaa4863a7bbf119f13e6983f396f9e10.png

Green is RSSI in MINUS dBm, i.e. it is about -30 dBm on the ground and about -60 dBm in the first few minutes of flight, which is totally fine. During the landing phase it is on long final and drops to -100 dBm but the link quality shows over 80% so most of the packets are still getting to the receiver in the plane, even though the flight controller thinks it is in throttle failsafe.
Note time is off in the Taranis log graph but the low RSSI is at the END of the flight, long after the failsafe and the RSSI is not low enough for a failsafe.
The FS occurs at 80 seconds after takeoff; the low but ok RSSI value in the Taranis log occurs when the plane is on long final for landing, many minutes after the failsafe event.

Why did Ardupilot call a failsafe event?
Why did it last several minutes through the whole flight, when the transmitter radio on the ground was in good contact with the radio receiver on the plane, as shown by the logs in the Taranis?
Why did the log file show 1500 for the rc3in value even though it was in failsafe?

As a bench test, here is the log when I have the plane on the bench and turn off the radio. It does what you would expect, i.e. the RSSI goes to zero and the throttle (RCIN3 raw) goes to 982, the failsafe value:
https://discuss.ardupilot.org/uploads/default/original/3X/1/b/1b8c08f167dd71a505a4e400804d919a59c68935.png

Benchtest (correct failsafe operation) log:
https://discuss.ardupilot.org/uploads/default/original/3X/a/e/ae1ec2c4b7ffebb181611dd7153a35f4fb651893.tlog

**Version**
Arduplane v 3.9.4 (Chibios)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Airframe: Nano Goblin 600 mm flying wing

**Hardware type**
Omnibus F4 Pro V3

**Logs**
Log during problem:
https://discuss.ardupilot.org/uploads/default/original/3X/e/2/e2de4f96445c41bb39de162da0afc92f96bc265d.BIN

https://drive.google.com/file/d/1V2jvoW2MNSqxyONqdYXMmztqNOC4cA4n/view?usp=drive_open

Log with bench test showing failsafe works on bench:
https://discuss.ardupilot.org/uploads/default/original/3X/a/e/ae1ec2c4b7ffebb181611dd7153a35f4fb651893.tlog

IamPete1, 2021-08-15:
Sorry we never got to this, is there still a issue on 4.1?

bgiannes123, 2021-08-16:
I do get this same issue on my pixhawk1 ‘copter’, 4.1, good Rssi/lq.  I’ve had it happen with FrSky Rx, and then again after I upgraded to tracer Rx same issue.  (My copter always rtl safely:)

andyp1per, 2021-08-16:
Tracer likely that you are using a UART w/o DMA

bgiannes123, 2021-08-16:
On a pixhawk1 aka flu-v2 serial2, 3, 4 are dma ports?   I’ll check and get back to you.

bgiannes123, 2021-08-16:
I can confirm it’s on serial 2 which is a dma port.  Is there anything I need to configure in MP to turn dma on?

andyp1per, 2021-08-16:
Sorry, don't know then

jywilson2, 2021-09-11:
I am seeing the same issue with ArduPlane for MatekSys H743 Wing firmware version 4.0.9.  In my case the "No RC Receiver" message occurs about three times during a 30 minute flight, but FS is *not* triggered during this time.

I am using a Frsky R9MM receiver.  It is connected via S.Bus and S.Port for pass through telemetry. 

Other than the No RC Receiver message the flight sequence is successful.  I am tempted to ignore the message, if it wasn't so ominous.