ArduPilot/ardupilot #12085: Copter: Misison Item in GuidedMode seems broken

tilaktilak, 2019-08-20:
### Bug report

**Issue details**

With the last version of sim_vehicle simulator from Master, the Mission item sent in guided mode to achieve GotoLocation (with QGC) does not work anymore. It works on last Copter-3.6 branch and tags 3.6.9 and 3.6.10.

Steps to reproduce :

-  `./sim_vehicle.py -w -v ArduCopter -S 3 --console -m --aircraft=copter`
- In QGC, ask for takeoff and send a GotoLocation point
- Error is shown as :
[image]

**Version**
Master : Hash => [5239d86d137fea828ca40fe1e23f033f127247b6](https://github.com/ArduPilot/ardupilot/commit/5239d86d137fea828ca40fe1e23f033f127247b6)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Default

**Hardware type**
Ardupilot Simulator

**Logs**
Changed the extension from tlog to log to attach it.
[flight.log](https://github.com/ArduPilot/ardupilot/files/3522583/flight.log)

peterbarker, 2019-08-21:
```
Got MAVLink msg: MISSION_ACK {target_system : 255, target_component : 0, type : 3, mission_type : 8}
```

Wow.  Mission type 8?

That's a new one on me.

It's not in our common.xml.  It's not in mavlink/mavlink as far as I can see (https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml)

@DonLakeFlyer what's up here?  This *used* to work...

I mean, if I'm guessing what's going on this could be a very nice way of doing things....

DonLakeFlyer, 2019-08-21:
Crap I see it now. Previsouly QGC+ArduPilot never switched to Mavlink v2 because of that negotiation problem. Now it is. And there is a bug in QGC where there is a typo in a memset call which sets the entire structure to 8! Which used to work fine since all the fields were filled in with real values for mavlink 1. But now with mavlink 2 the mission type is left at 8. Memset has been wrong for years now! But just finally git me in the ass now. Sorry about that.

tilaktilak, 2019-08-21:
Great we found the bug, I will close this issue as it's not related to ArduPilot then.
@DonLakeFlyer if it's not already done I can create a PR in QGC to fix the memset ? 
Thanks guy !

kkroeger007, 2019-08-22:
I can confirm I had a similar issue previously opened on the forums just a few days prior as well. It was the mavlink versioning we were using against the latest arducopter branch of code. This should not be an issue within the arducopter branch of code and should be closed.

closed 2019-08-22 by tilaktilak