ArduPilot/ardupilot #12109: Copter: Add Flip After Crash Mode, "aka Turtle Mode"

jcarff, 2019-08-21:
This is a new flight mode request. I am hoping that a common and useful feature in betaflight firmware can be brought over to arudpilot: "Flip After Crash Mode". This has been a useful feature for me in my own personal drones, enabling me to more easily retrieve a drone that has gotten stuck in an area that is not easily or safely accessible by foot. My main reason for requesting it now is i am currently working on developing drones that can fly through and search buildings in dangerous situations, For example if there is an active shooter in a building, a chemical spill, or a possibility of structural collapse. The arudupilot enabled flight controllers more easily allow me to integrate various sensors that make it much easier for a novice pilot to navigate in these enclosed spaces, but inevitably there is always a chance of crashing. If a crash does take place, and the drone lands on its back, I would like the option for the operator to be able to flip the drone into "flip after crash" mode and flip the drone over to enable to mission to continue instead of just ending there.

rmackay9, 2019-08-22:
Tridge and I discussed this mode a bit in the past as well.  I think It requires spinning at least some of the motors in reverse which is possible with some ESCs including DShot enabled ESCs.

jcarff, 2019-08-22:
Yes this does require spinning the props in reverse and it works very well on our current betaflight drones. Is reversing the motors something possible with Aurdupilot and DShot enabled ESCs. Not only would this allow for flipping after crashing but also allow for 3D drone flights as well which I have seen a few others asking for. I am very motivated to get this to happen :)

jcarff, 2019-08-23:
If there is a way to figure out motor reversing, I can probably get a couple developers on getting the flight mode (flip) coded up.

auturgy, 2019-08-23:
FYI Motor reversing has been implemented on DShot esc’s for plane (https://github.com/ArduPilot/ardupilot/pull/9743) but not pulled across into copter. Randy is definitely the best to advise on how to do that, but the pr I linked to will give an idea of hooking into the library.
The “turtle mode” I’m guessing just picks two motors based on attitude, and gives them a bit of reverse power until attitude is beyond vertical.

Regards,

James

jcarff, 2019-08-24:
>  The “turtle mode” I’m guessing just picks two motors based on attitude, and gives them a bit of reverse power until attitude is beyond vertical. Regards, James

It is actually super easy barely an inconvenience, no real logic required, once you switch to turtle mode, the motor directions reverses but the props do not spin until you add input. What ever way you push your roll/pitch stick, the 2 props required to flip the drone in that direction throttle up. Thats it. 

I will pull the code first chance I get to check out the motor reversing but any help I can get in bringing this over to copter would be greatly appreciated. I am at least hopeful knowing now that it is possible. I wouldnt be able to switch our quads over to aurducopter for this next project coming up if it wasn't possible.  :)

jcarff, 2019-09-04:
@rmackay9 I was wondering if you can give me some guidance on this. Im looking at the reverse thrust implementation for arduplane [(#9743)](https://github.com/ArduPilot/ardupilot/pull/9743) and was wondering if you think there would be any issues in bringing this over to arducopter.  When I change to FlipOver mode I simply need the motors to stop spinning, motor direction reversed, and then stick input commands the throttle of the 2 motors in direction the stick is pushed flipping the drone in that direction. no automated logic involved.

rmackay9, 2019-09-05:
@jcarff, that might be OK as a first step but longer term I think it would be good if we tried to do some attitude control.  So the mode would be a bit more like our existing Flip or Throw modes where we monitor the attitude and try and correct it.

jcarff, 2019-09-06:
@rmackay9 I was wondering if you could point me in the right direction, I have a new flight mode, FlipAfterCrash, added and setup in [my own branch](https://github.com/jcarff/ardupilot/tree/CopterFlipAfterCrash) and now I am looking through how to not only command the motors to reverse, but mainly how to not use any stabilization and just control the motors directly.

How I want the procedure to work for when I switch to this new flight mode:
-After you crash and are upside down-
1)You must first disarm the drone,
2)Switch to flip after crash flight mode
3)Arm the drone (i need to stop the props from spinning automatically)
4)Push the stick in the direction you want to flip
     - motor direction is reversed
     -you have 4 flip directions, at no point will more than 2 motors spin at a time, 
     - motor speed should be directly controlled by how far you have moved the stick in that direction
5) Disarm the drone
6) Change back to your flight mode
7) Arm and continue the flight.

jcarff, 2019-09-30:
@rmackay9 a small bit of guidance might help me finish this off. I now have my own flight mode added that lets me directly control the appropriate motors depending on stick input and now i am trying to actually get things reversed. I used blhelisuite to set the esc to bidirectional but I am not sure what on the arducopter side of things I need to change. For reverse thrust on arduplane one of the instructions is to "set param THR_MIN to a negative number (maximum amount of reverse thrust you want, -100 means 100%)"  but I am not sure if there is an equivalent to THR_MIN for copter.  now that I have bidirecitonal enabled I am not sure what other copter parameters need to change to make this functional. Any guidance would be greatly appreciated.

jcarff, 2019-10-01:
@tridge you might actually be able to help me with this since you implemented it for plane. I think I am close, but I know I am missing something. 
I have a new flight mode added that lets me directly control the appropriate motors depending on stick input and now i am trying to actually get things reversed.

I have setup my speed controllers using blhelisuite  to be bidirectional. 

I have SERVO_BLH_AUTO set to 1.

so i need to set SERVO_BLH_MASK since SERVO_BLH_AUTO is 1

what do i need to set SERVO_BLH_REMASK to?

there is no "THR_MIN" in arducopter, do you know an an equivalent that i need to set to -100 so that my throttle range is correct.  
Thank you

jcarff, 2019-10-04:
@tridge and @rmackay9, i know you are busy but I hope you can find a moment to help me out, this is my one last plea for help. :( if i cant figure this out in about 2 weeks I have to ditch my plans to use ardupilot on these drones. Here is what I am at right now. Ignoring any flip after crash modes, All I want to be able to do is setup my BLHeli esc to be bidirecitonal and then set "servo_blh_remask" to 15 (my 4 motors),  and still be able to fly in normal modes (no reversing). At the moment, if I set "servo_blh_remask" to 15, when i am in stabilize mode and try to arm, the arm command is accepted, but no motors spin up, once i bring the throttle up to 50% one or two of my motors arm, and if i try and disarm, those motors rev up to maximum and do not turn off unless i unplug the battery.

rmackay9, 2019-10-05:
@jcarff, re trying to get the BLHeli ESCs working as a normal setup (i.e. no reversing, etc) it's probably best to look at [this wiki page](http://ardupilot.org/copter/docs/common-dshot.html) (which you've probably already seen) or [check in the forums](https://discuss.ardupilot.org/c/arducopter/copter-36).

There's also some more discussion going on in [this issue](https://github.com/ArduPilot/ardupilot/issues/12494).

I've never personally used BLHeli ESCs on a copter so I'm afraid I'm not going to be of much help.

snakeeater12, 2019-11-20:
Man having this function would make life easier.

Can't we just copy the code from Beta flight?

Sebbb, 2021-06-20:
Now with 4.1 and "Support dshot commands for LED, buzzer and **direction control**", isn't it something that should be easy to implement on top?

andyp1per, 2021-06-20:
Yes should be easy now

khancyr, 2021-06-20:
What prevent us to use it with normal ESC ? That is just pushing in one direction now ?

andyp1per, 2021-06-20:
You have to be able to reverse the motor direction temporarily

andyp1per, 2021-06-20:
@jcarff if you rebase your branch I can help you finish this off

IamPete1, 2021-08-14:
Closed by https://github.com/ArduPilot/ardupilot/pull/18140

closed 2021-08-14 by IamPete1