ArduPilot/ardupilot #12116: Copter: Matek F405 Motor Order

DaneSlattery, 2019-08-22:
**Issue details**
Just had my Maiden Flight with a Matek F405-CTR Flight Controller.

Having double-checked my wiring, I am sure of a bug with the Motor Ordering. 

I had to use Mission Planner to change the Outputs from 
1 -> 1
2 -> 2
3 -> 3
4 -> 4

TO 
1 -> 4
2 -> 1
3 -> 2
4 -> 3

I realised the bug when trying to fly for the first time, and having the stabilise, but when trying to correct pitch, it would continue to pitch more and more until crashing. Luckily I was tethered.

**Version**
Firmware 3.6.9

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X configured Quad

**Hardware type**
Matek F405

**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2019-08-22:
I'm fairlty sure the pwm outs match the silk screen, I think selecting betaflight x for the [frame type](http://ardupilot.org/copter/docs/parameters.html#frame-type-frame-type-x-v-etc) would have avoid having to change the pwm ordering. In any-case you should always run a motor test to double check the ordering is correct before your first flight.

closed 2019-08-23 by IamPete1

DaneSlattery, 2019-08-25:
Perhaps, but at least by having the issue, someone else can be aware of the required configuration. That BetaFlight-X is unfortunately not documented here: http://ardupilot.org/copter/docs/connect-escs-and-motors.html#connect-escs-and-motors3, so I didn't know about it.

Thank you, and yes, these were observations from my motor test.

jimpriest, 2019-11-30:
I'm on 3.6.11 and trying frame_type = 12 (betaflight) but that then leads to a arming error:  PreArm: check firmware or frame_class.  I have frame_class = 1 (x quad).

IamPete1, 2019-11-30:
@jimpriest betaflight x doesn't exist in 3.6 its in 4.0 and newer, you can always use normal x and change the servo outputs to suit