ArduPilot/ardupilot #12123: NotchFilter: INS_NOTCH_FREQ = 10 causes core dump in SITL

rmackay9, 2019-08-23:
During SITL testing of the new harmonic filter I needed to test parameter conversion so I changed INS_NOTCH_FREQ from the default of 80 to 10.  This caused a coredump in SITL on startup which meant the only way to recover was to remove the eeprom.bin (i.e clear all parameters).  The value of "10" is apparently in the acceptable range of values according to the NotchFilter.cpp's FREQ parameter description.

I'm unsure of the seriousness of the issue if this is done on a real board (SITL is purposely more sensitive so we can catch errors).

rmackay9, 2019-08-23:
By the way, this bug is fixed by this PR from @andyp1per and @lthall: https://github.com/ArduPilot/ardupilot/pull/11795

andyp1per, 2019-08-23:
@rmackay9 - yes I noticed this when coding the harmonic changes - basically if your notch center frequency is less than half the notch bandwidth (which I am assuming is the case here) then the whole thing blows up

rmackay9, 2019-08-23:
@andyp1per, ok great.  we can close this issue once your PR goes in.

amilcarlucas, 2019-09-29:
#11795 is in. Can this be closed ?

rmackay9, 2019-09-29:
yes, thanks!

closed 2019-09-29 by rmackay9