ArduPilot/ardupilot #12129: SPM4649T

Bumerangas, 2019-08-24:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**
 problem with mRo x2.1 simple or 777 version with DSMX receivers ( Spektrum SPM4649T or OrangeRx R111XN ) for very long time and still can`t use any of them, binding ok but channels not moving.

Checked all latest and older arducopter pilot rover firmwares, no lucky

SPM4649T looks like SRXL receiver, so powered from separate power pin 5V and anyway not working.

**Version**
all versions

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad

**Hardware type**
mRo x2.1 simple or 777 

**Logs**
-

Bumerangas, 2019-08-24:
this receivers work very well with betaflight, this firmware have separate sepktrum satellite and spektrum SRXL protocols chose.

Naterater, 2019-09-04:
Similar issue with SPM9745 on cube blue with mini carrier board set to 3.3v and DSM on the solder bridges.

cglusky, 2020-02-08:
I can not get Frsky X8R or R-XSR to show up in Mission Planner radio calibration either.  I can switch them to a PixRacer and they work fine.  It's just mRo x2.1.

BUT a XSR works just fine.

So I wonder if this needs a better title???