ArduPilot/ardupilot #12132: SITL floating point expection on watchdog lockup

Jaaaky, 2019-08-25:
### Bug report

**Issue details**

I was testing some watchdog stuff for pr #12131 but sending `lockup_autopilot IREALLYMEANIT` to SITL caused recurrent floating point exception and crash every-time.
 Here is the trace I got on terminal;

```
0x00007f87f05ade0a in waitpid () from /lib64/libc.so.6
#0  0x00007f87f05ade0a in waitpid () from /lib64/libc.so.6
No symbol table info available.
#1  0x00007f87f052b4af in do_system () from /lib64/libc.so.6
No symbol table info available.
#2  0x00000000004b57db in AP_HAL::dump_stack_trace() ()
No symbol table info available.
#3  0x00000000004b5c61 in _sig_fpe(int) ()
No symbol table info available.
#4  <signal handler called>
No symbol table info available.
#5  0x00000000004271d6 in AP_Baro::get_EAS2TAS() ()
No symbol table info available.
#6  0x000000000048e578 in AP_RollController::_get_rate_out(float, float, bool) ()
No symbol table info available.
#7  0x0000000000406e76 in Plane::stabilize() ()
No symbol table info available.
#8  0x000000000045953d in AP_Scheduler::run(unsigned int) ()
No symbol table info available.
#9  0x00000000004597a1 in AP_Scheduler::loop() ()
No symbol table info available.
#10 0x0000000000405110 in Plane::loop() ()
No symbol table info available.
#11 0x00000000004ad6da in HAL_SITL::run(int, char* const*, AP_HAL::HAL::Callbacks*) const ()
No symbol table info available.
#12 0x0000000000405d0c in main ()
No symbol table info available.

Thread 1 (Thread 0x7f87f1587740 (LWP 7781)):
#0  0x00007f87f05ade0a in waitpid () from /lib64/libc.so.6
No symbol table info available.
#1  0x00007f87f052b4af in do_system () from /lib64/libc.so.6
No symbol table info available.
#2  0x00000000004b57db in AP_HAL::dump_stack_trace() ()
No symbol table info available.
#3  0x00000000004b5c61 in _sig_fpe(int) ()
No symbol table info available.
#4  <signal handler called>
No symbol table info available.
#5  0x00000000004271d6 in AP_Baro::get_EAS2TAS() ()
No symbol table info available.
#6  0x000000000048e578 in AP_RollController::_get_rate_out(float, float, bool) ()
No symbol table info available.
#7  0x0000000000406e76 in Plane::stabilize() ()
No symbol table info available.
#8  0x000000000045953d in AP_Scheduler::run(unsigned int) ()
No symbol table info available.
#9  0x00000000004597a1 in AP_Scheduler::loop() ()
No symbol table info available.
#10 0x0000000000405110 in Plane::loop() ()
No symbol table info available.
#11 0x00000000004ad6da in HAL_SITL::run(int, char* const*, AP_HAL::HAL::Callbacks*) const ()
No symbol table info available.
#12 0x0000000000405d0c in main ()
No symbol table info available.
A debugging session is active.
```

**Version**
_Arduplane master_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Logs**
Tell me what is needed

Jaaaky, 2019-08-25:
I don't know if this can also affect real vehicles.

Jaaaky, 2019-08-25:
I could apparently solve the problem by the following patch

```
diff --git a/libraries/AP_Baro/AP_Baro.cpp b/libraries/AP_Baro/AP_Baro.cpp
index 50eed968f0..ea1e52506f 100644
--- a/libraries/AP_Baro/AP_Baro.cpp
+++ b/libraries/AP_Baro/AP_Baro.cpp
@@ -313,6 +313,9 @@ float AP_Baro::get_altitude_difference(float base_pressure, float pressure) cons
 // assumes standard atmosphere lapse rate
 float AP_Baro::get_EAS2TAS(void)
 {
+    if (hal.util->was_watchdog_reset() && AP_HAL::millis() < 10000) {
+        return 1.0f;
+    }
     float altitude = get_altitude();
     if ((fabsf(altitude - _last_altitude_EAS2TAS) < 25.0f) && !is_zero(_EAS2TAS)) {
         // not enough change to require re-calculating
```

If you think it's a valid solution, I can send a pr

peterbarker, 2019-08-25:
On Sun, 25 Aug 2019, Jaaaky wrote:

> I could apparently solve the problem by the following patch

We should work out what the issue is here - could you compile with debug 
symbols and have a look, please?

I'm guessing get_pressure(...) is zero.

> If you think it's a valid solution, I can send a pr

It *may* be better to check get_pressure in case this can happen when we 
haven't had a watchdog reset :-)

Jaaaky, 2019-08-25:
@peterbarker
Yeah, that's it. Can you check this pr #12135 ?

PR #12135 (2019-08-26): AP_Baro: Fix floating point exception with watchdog reset
Fixes #12132
commit fee3136b (2019-08-26): AP_Baro: Fix floating point exception with watchdog reset

Fixes #12132
closed 2019-08-26 by tridge