ArduPilot/ardupilot #12140: SITL mis-attitude

Jaaaky, 2019-08-26:
### Bug report

**Issue details**

I'm debugging a stall -on loiter/turns- bug found on ArduPlane master.
I'm reproducing the bug using SITL, while doing so I found another SITL bug using `git bisect`.
When on loiter plane suddenly have very bad attitude, I've attached 2 log files, one showing the bug caused by commit a84af856896980e90ecee08ca541bc005c97cf1a . Another one with my patch applied

Here is the patch:
```
diff --git a/libraries/AP_InertialSensor/AP_InertialSensor_SITL.cpp b/libraries/AP_InertialSensor/AP_InertialSensor_SITL.cpp
index f187723d71..4f830ef56d 100644
--- a/libraries/AP_InertialSensor/AP_InertialSensor_SITL.cpp
+++ b/libraries/AP_InertialSensor/AP_InertialSensor_SITL.cpp
@@ -118,7 +118,7 @@ void AP_InertialSensor_SITL::generate_accel(uint8_t instance)
 
     uint8_t nsamples = enable_fast_sampling(accel_instance[instance])?4:1;
     for (uint8_t i=0; i<nsamples; i++) {
-        _notify_new_accel_raw_sample(accel_instance[instance], accel);
+        _notify_new_accel_raw_sample(accel_instance[instance], accel, AP_HAL::micros64());
     }
 }
 
@@ -163,7 +163,7 @@ void AP_InertialSensor_SITL::generate_gyro(uint8_t instance)
     
     uint8_t nsamples = enable_fast_sampling(gyro_instance[instance])?8:1;
     for (uint8_t i=0; i<nsamples; i++) {
-        _notify_new_gyro_raw_sample(gyro_instance[instance], gyro);
+        _notify_new_gyro_raw_sample(gyro_instance[instance], gyro, AP_HAL::micros64());
     }
 }
 
```

The bug is reproducible with xplane on more than 90% of trials within the 1st two loiter rounds.

**NOTE: I'm pretty sure that there is a similar issue with real vehicles on master, but I don't know how can I prove that. Maybe the same exact problem or another problem with inertial sensors recent changes, so I hope you investigate this too.**

**Version**
master bisected at commit a84af856896980e90ecee08ca541bc005c97cf1a 

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Logs**
_Please provide a link to any relevant logs that show the issue_
[attitudelossbug.zip](https://github.com/ArduPilot/ardupilot/files/3541531/attitudelossbug.zip)

Jaaaky, 2019-08-27:
Could be an `if nsmaples == 1 ... else ...` ?

tridge, 2019-08-27:
interesting!
note that XPlane normally uses EKF type 10 to avoid issues like this. We don't have lock-step scheduling for xplane, so timing is always problematic.
This change will actually create errors on sim backends other than xplane, as it puts the wrong timestamps on IMU samples (when nsamples if more than 1 it will set the same timestamp on all samples, which is incorrect)

Jaaaky, 2019-09-06:
@tridge
Any updates here?

If busy I can send a pr with `if nsmaples == 1 ... else ...` ?

IamPete1, 2021-08-15:
I sounds like the solution's is not to run the EFK with with none lock-step SITL backends.

closed 2021-08-15 by IamPete1