ArduPilot/ardupilot #12164: Smart RTL return to Rally Points as well

Autonomost, 2019-08-28:
### Feature request

**Is your feature request related to a problem? Please describe.**
When Copter engages smart RTL, it won't return to Rally Points and land, only the home point. 

**Describe the solution you'd like**
I would like the Copter to return to Rally points as well as the Home Point

**Describe alternatives you've considered**
I have scoured the parameters and haven't found anything to make the copter fly to a rally. Unless I switch to regular RTL. 

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

peterbarker, 2019-08-28:
On Wed, 28 Aug 2019, Autonomost wrote:

> When Copter engages smart RTL, it won't return to Rally Points and land, only the home point.

SmartRTL won't have a path to any Rally Point.  How would it know how to 
get to one?

One option might be to have a parameter saying if we get within so many 
metres of a Rally Point point when flying the return path then pop over to 
that Point.

rmackay9, 2019-08-28:
unless there are ideas on how we could do this it's tempting to close it.  I've also thought that it would be nice if we supported Rally Points but realistically I'm not sure we have a good way to do it.

lvale, 2019-08-28:
Unless a Rally Point happens to coincide with a saved point (which is extremely unlikely/approaching impossible) there's no way to achieve this, without resorting to Path Planning and Collision avoidance, so I agree to close it.

tobymoores, 2019-08-30:
Might it be useful to designate selected mission waypoints as rally points?

tobymoores, 2019-08-30:
We have a requirement to fly close to structures at a distance. Smart RTL works really well as it ensures that the shortest path RTL is not though said structure. There are instances however when a closer rally point would be a better option. This waypoints as rally points for smart RTL

Pedals2Paddles, 2019-08-30:
The only way this could work is if the rally point is laterally within X meters of the SmartRTL breadcrumb path _AND the operator wants to accept the risk of hitting whatever may be vertically between the copter and the ground_.  Since the copter has not flown through the vertical airspace between it and the rally point, it can't know it is a safe, clear path.  So that would be a risk for the operator to take.  

That being said, I can't imagine why an operator would put a rally point in a location that isn't clear all around. So the risk is probably pretty low actually.  However, it is a risk that we can't mitigate in code.  I'm not sure it is worth the complication even if it is low risk.

tobymoores, 2019-08-30:
Agreed. That’s why I think it has to be a waypoint (or breadcrumb). Still useful though if the copter is range limited

Pedals2Paddles, 2019-08-30:
> Agreed. That’s why I think it has to be a waypoint (or breadcrumb). Still useful though if the copter is range limited

So what you're describing would only apply in a mission, right? And somehow a waypoint would need to also be tagged as a rally point.

peterbarker, 2019-08-30:
On Fri, 30 Aug 2019, Matt wrote:

>
>       Agreed. That’s why I think it has to be a waypoint (or
>       breadcrumb). Still useful though if the copter is range
>       limited
> 
> So what you're describing would only apply in a mission, right? And
> somehow a waypoint would need to also be tagged as a rally point.

If missions are being used then the existing DO_LAND_START logic would 
work quite nicely, I'd think.

This isn't SmartRTL - this is the operator deciding exactly what the 
vehicle should do if it must RTL in specific places.  SmartRTL is NOT 
involved.

Currently only in-tree for Plane.

Copter has an open PR here for it:

https://github.com/ArduPilot/ardupilot/pull/11553/commits

rmackay9, 2019-08-30:
I've been thinking about writing an MRTL mode (for Mission RTL) which would replace the PR that @peterbarker has linked,

When engaged it would first look for the closest point on any segment in the mission and then follow the segments back to home.  A "segment" is the line from any waypoint command to the next waypoint command.  The tricky part is building up the segments taking into account do-jump commands but it's very do-able.

tobymoores, 2019-08-31:
Apologies. I am confusing waypoints and breadcrumbs. Here are some examples which might be helpful:

1. Smart RTL but stop safe distance from launch point
follow the breadcrumbs all the way home but come to hover x meters from launch then apply preselected secondary action for example:
Wait to re-establish link then after delay/batt minimum
Land
Activate chute
RTL

2. Smart RTL until safe distance from known structure
follow the breadcrumbs until x meters from structure then apply secondary action as above

3. Smart RTL until adjacent to a rally point
follow the breadcrumbs until as close as possible to a rally point without leaving the pathway then apply secondary action as above

4. Smart RTL until intersection with preplanned emergency route
follow the breadcrumbs until an emergency route crosses its path. Switch to auto and fly mission to land.

5. Smart RTL and return to mission
assume I have left a mission to investigate something that would require Smart RTL (SAR grid search is paused to allow me to search a building or wood of interest)
Follow the breadcrumbs until I rejoin the original mission path. Resume mission from there

lvale, 2019-08-31:
@tobymoores Have you tried to do your use cases using the new Lua script functions ?

tobymoores, 2019-08-31:
We’ve done some similar work using basic rally points but nothing as complex as this.

I’m a big fan of Smart RTL and I think the simplicity of the breadcrumb approach is a really powerful thing.

My preference is to have safety features like this in the core code so everybody benefits rather than in scripts or on the CC. My scenarios are really anticipating EVLOS and BVLOS situations that will become increasingly commonplace

This is my 1st dive in to a features request so apologies for adding noise to the signal.

proficnc, 2019-08-31:
I do Iike the pre planned emergency route idea.

It would lay down prepared breadcrumbs as a safe path, and if SRTL found them as a path, it could use them. 
A radius of pre determined “safety “ around a rally point could do the same thing if it happened to intersect with the breadcrumbs etc...

tobymoores, 2019-09-01:
@proficnc yeah I like that. I wondered about a SRTL triggered safety fence (think geofence or no fly zone) where SRTL would follow the breadcrumbs until it cleared the fence but would then be free to RTL direct or some other action.

Crossing or approaching other fences might cause or inhibit further actions.

I guess my scenarios 1, 2 and 3 would be covered by this. Very cool!

rmackay9, 2019-09-02:
@lvale, I agree that once it becomes too bespoke it should go into lua scripts.  Our current lua script feature can't control vehicles movement (yet).  it's only for auxiliary functions so far.

lvale, 2019-09-03:
@rmackay9 exactly why I've pinged @WickedShell about the "need" to extend the Lua framework ;)

IamPete1, 2021-08-15:
Closed by https://github.com/ArduPilot/ardupilot/pull/18075 it allows multiple return options as programmed in a mission.

closed 2021-08-15 by IamPete1