ArduPilot/ardupilot #12165: CAN pain points and how to resolve them

tridge, 2019-08-29:
This is to discuss some current pain points for use of CAN in ArduPilot and how to resolve them

 - DNA requires sdcard, which may not be available (especially as we now support microsd insert after boot)
 - slow CAN peripheral boot causes issues with sensor detection (especially mag, baro and airspeed)
 - no support for dual CAN GPS
 - no support for RTK correction injection with CAN GPS
 - no support for dynamic discovery of new devices (especially mag)
 - current stable fw releases have no SLCAN and no DNA

proficnc, 2019-08-29:
-device priority should be selectable by device ID
Ie, parameters that list the device ID of each sensor, and remain consistent each reboot till purposely changed. 
-a way of setting up device priorities for a new build.

proficnc, 2019-08-29:
- devices need to be kept track of by their hardware UID
- device discovery should happen continuously until all devices are discovered.

tridge, 2019-08-29:
some ideas that Randy and I discussed recently:
 - could have COMPASS_TYPE, COMPASS_TYPE2, COMPASS_TYPE3, which default to AUTO. If not AUTO then specifies compass type, eg. CAN, HMC-I2C etc. This would allow uses to say they want CAN compass first. Possibly auto-set these parameters on calibration if not already set?
 - could automatically re-order compass devices array to line up detected devices with compass ID parameters, so that if detection order changes than device list is fixed to match parameters
 - should we switch to using uart-over-can for GPS, so that RTK works and all other GPS driver features work?

tridge, 2019-08-29:
for DNA, we need to work out how much space we need in StorageManager. We can potentially reclaim space currently in 16k hal.storage by allowing for waypoints, fence etc to be stored on microSD

rmackay9, 2019-08-29:
@tridge, maybe the _TYPE parameter should be a desired DEVID which includes the device, port and protocol.

tridge, 2019-08-29:
@rmackay9 the problem with the devid is it is such a difficult number for a user to understand as it is made up of bitfields

bugobliterator, 2019-08-29:
@tridge Maybe a user interface can be made to resolve that issue. In Mission Planner and MAVProxy to start with.

tridge, 2019-08-29:
> @tridge Maybe a user interface can be made to resolve that issue. In Mission Planner and MAVProxy to start with.

which issue? (there are so many!)

bugobliterator, 2019-08-29:
> > @tridge Maybe a user interface can be made to resolve that issue. In Mission Planner and MAVProxy to start with.
> 
> which issue? (there are so many!)

Sorry, the _TYPE parameter being a desired DEVID

tridge, 2019-08-29:
corrupted CAN devids fixed with this PR: #12168 
@rmackay9 we need that fix for stable releases too

Evan1010, 2019-08-29:
I would also like to point out that Dual UAVCAN Airspeed is perhaps defunct, per my testing of Master. If two sensors are attached, it will just grab the first one it sees and run with it for both enabled sensors (set to UAVCAN type). Notable that UAVCAN airspeed support is not yet in a stable release.

This however may be a symptom of something other than UAVCAN implementation, as this lack of dual sensor support seemingly applies to any two identical sensors. See relevant links:

[Discuss - Dual SDP3x](https://discuss.ardupilot.org/t/connecting-two-sdp31-pressure-sensors-for-dual-airspeed-not-working/37629)

[Issue Report - Dual SDP3x](https://github.com/ArduPilot/ardupilot/issues/10305)

[Issue Report - Dual MS4525DSO](https://github.com/ArduPilot/ardupilot/issues/11880)

Additionally, regarding Dual UAVCAN GPS support: I currently have an aircraft using dual Here 2 GPS units on a single UAVCAN bus - I _appear_ to get independent and proper readings from both of them (Blending is disabled). I do get "Bad GPS Signal Health" rather often, for what it is worth.

If this is actually broken and I've been fooling myself this whole time, I'd like to know so that I can take appropriate action for reliability's sake. Same for the magnetometers onboard the Here 2 units . . . I had thought they were both working properly.

WickedShell, 2019-08-29:
@Evan1010 dual DLVR, and dual MS5525 both work. (If you set the addresses on the 5525 you can even do dual sensors on the same I2C bus).

Evan1010, 2019-08-29:
@WickedShell Thank you for the correction.

It seems just my ~luck~ ignorance to try the three (UAVCAN, MS4525, SDP33) that don't currently support dual sensors.

Is dual UAVCAN airspeed still something worth adding to this list? Airspeed support for UAVCAN isn't released outside of Master yet, so now might be a good opportunity without worrying about breaking existing implementations.
Options otherwise are I2C transport over UAVCAN (MS5525/DLVR), dual DLVR over I2C, or dual MS5525 over I2C. 
The first two options also require new major release(s) for feature support - DLVR driver in release, and AP_Periph development respectively. I recall some issues with the MS5525 regarding I2C integrity, noise and offset that turned me away from them.

WickedShell, 2019-08-29:
@Evan1010 I've had issues with everything except DLVR thus far that's turned me off of them. (As a caveat I'm the one who did the initial DLVR driver as I was moving on to trying other sensors after having aircraft loss/near aircraft loss events with all the other sensors we support, so I may be a bit biased).

proficnc, 2019-08-29:
Multiple CAN airspeed is definitely on the list of wanted things

tridge, 2019-10-18:
new DNA server is merged, thanks Sid!
I'm going to work on RTCM injection support this week

rmackay9, 2020-02-20:
The issue listed above, "no support for dual CAN GPS" is resolved?  An issue came up today in the Copter-4.0 forums with a user hitting an "Internal Error 0x400" (constrained NaN) when using dual UAVCAN GPSs.  [The discussion is here including a log](https://discuss.ardupilot.org/t/copter-4-0-0-released/50545/79).

IamPete1, 2021-08-15:
I think we could probably close this, is there is anything outstanding we might be better off tracking in a new issue.