ArduPilot/ardupilot #12170: Plane: loitering for climbing first during RTL

robustini, 2019-08-29:
### Feature request

I had a doubt about using SOAR feature on a slope, so flying with a glider under takeoff height, the problem that could arise is the automatic RTL when the glider is below the takeoff height and returning to the home could crash into the slope if the rate of climb does not exceed the proximity to the mountain.
So what's missing in Plane (I think)?
An RTL that first reaches a certain height, for example in a sorta of Loiter, then begins go to the home point, as with the Copter.
This new function should obviously be an optional, even deactivated.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

Redshifft, 2019-08-30:
Plane must have an option to climb for a few seconds before it has a chance to bank straight into the ground on RTH or failsafe RTH, Unsure why so few understand that around here :(

robustini, 2019-08-31:
Very often the most logical things are not taken into consideration because you don't need them, but this is a necessary feature in many cases.
We can solve the problem by using rally points but it becomes more laborious, my proposal in many cases could already make them useless.

Redshifft, 2019-08-31:
Yes, the basic problem is people traditionally don't setup and test Failsafes like other flight functions, if they did this would have been resolved way back :)
I posted this in Chibios a couple of weeks back.. 
"@tridge did you ever get a chance to add that climb before turn option to the RTH that you were going to look into ? I have another idea which would work for low altitude failsafe scenarios, that would be simply adding an extra option to FS_SHORT_ACTN to have a "climb". this would run for FS_SHORT duration then RTH in FS long could not turn the plane into the ground. What do you think?"

samuelctabor, 2019-08-31:
There's actually a few options.

1) RTL along a slope to home and RTL altitude (current)
2) climb at maximum climb rate while heading towards home, until RTL altitude or some fraction of it is reached 
3) gain height while circling up to RTL altitude, or some fraction of RTL altitude, before continuing

1) is most conservative of battery whereas 3) is safest to avoid terrain.

I'm not sure if the terrain following feature could be used to add safety with the current behaviour? Will a spiral climb be used if the terrain is steeper than maximum climb rate permits?

robustini, 2019-08-31:
About option 3, the terrain follow would be useless if the climb rate does not allow you to climb so fast to avoid a collision.
So the spiral climbing at the declared RTL altitude is the best solution.
The battery failsafe should be properly set considering this climb, I do not see the problem.

Redshifft, 2019-08-31:
I know little about the SOAR function but do know it's unlikely to find it's way out of a tricky situation in the mountains, only the pilot could make the best decisions. So as with any craft if you blank the Video or Radio a wings level climb is the only solution to hopefully re-link your Video or RC.
I worry about any automatic battery failsafe which would take control away from the pilot - OK for hovering a multi rotor around a field but personally I would sooner destroy  a battery than a Model and choose where it crashes/lands

samuelctabor, 2019-08-31:
Maybe a bit in FLIGHT_OPTIONS could be used to select this climb-first option.

In general I think the number of parameters is slightly out of control. For example recently there were all the parameters for 10 rangefinders in the master branch. How many users can really be using this many rangefinders? All users bear the burden of downloading these parameters every time they connect. At least the number of rangefinders should be specified and only the corresponding parameters sent.

robustini, 2019-08-31:
@Redshifft it can happen that you are under the slope in the mountains of about fifty meters looking for thermals with the SOAR and you are very close to the mountain, you engage RTL and can be disastrous.

Redshifft, 2019-08-31:
@robustini I've done plenty of slope and ridge soaring, you hug the contours very closely when "scratching" when the lift goes !  I don't see any way an autopilot is going to always do the right thing in Failsafe. of course we were real men doing it without motors :D :D :D and a big straight 500mtr drop into the sea below gets your attention.
I would imaging the SOAR thing is more about floating about sniffing thermals than playing around in treacherous mountain winds ?

robustini, 2021-03-17:
Are [these commits](https://github.com/ArduPilot/ardupilot/commit/29043d62e9be8214f4232b6310699ff868ac7934) inherent to this feature?
In case I close the request and thank you!