ArduPilot/ardupilot #12171: Questions on the compass calibration function of "thin_samples()"

AnakinQin, 2019-08-29:
#12156 # Bug report

**Issue details**

The thin_samples() function in "CompassCalibrator.cpp"  would shuffle the total sample and judge each sample[i] to be acceptable by function "accept_sample()". However in this "accept_sample()" function, ergodicity of the search in _samples_collected array is done when calculate the distance between each sample as we can see I pasted below.
bool CompassCalibrator::accept_sample(const Vector3f& sample)
{
    static const uint16_t faces = (2 * COMPASS_CAL_NUM_SAMPLES - 4);
    static const float a = (4.0f * M_PI / (3.0f * faces)) + M_PI / 3.0f;
    static const float theta = 0.5f * acosf(cosf(a) / (1.0f - cosf(a)));

    if(_sample_buffer == nullptr) {
        return false;
    }

    float min_distance = _params.radius * 2*sinf(theta/2);

    for (uint16_t i = 0; i<_samples_collected; i++){
        float distance = (sample - _sample_buffer[i].get()).length();
        if(distance < min_distance) {
            return false;
        }
    }
    return true;
}

So, this process would possibly kill all the samples which we got and yet done ten times run_sphere_fit() calculation in state "COMPASS_CAL_RUNNING_STEP_ONE". When we enter "COMPASS_CAL_RUNNING_STEP_TWO", we might face an empty sample buffer then and have to re-collect the all 300 samples again.

Would anyone help me understand this design logic? Why shall we abandon them befor we go to "COMPASS_CAL_RUNNING_STEP_TWO"?

**Version**
whatever version it is

**Platform**
[ √] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
hex

**Hardware type**
Pixhawk

**Logs**
none

AnakinQin, 2019-08-29:
Should we leave the sample itself when do distacn judge in "accept_sample()" function?
If it's possible to do some change, how can I take my first commit to this project, I must be eager to do so.

AnakinQin, 2019-08-29:
@tridge I looked through the history and found you committed this part 5 years ago. Would help me explain the meaning of this part?

AnakinQin, 2019-08-29:
@tridge 
I looked through the history and found you committed this part 5 years ago. Would you kindly help me explain the meaning of this part? Thank you!

peterbarker, 2020-01-07:
Thanks @lukezhqin  - your PR helped us get that bug fixed.

closed 2020-01-07 by peterbarker