ArduPilot/ardupilot #12188: Logging: log used IMU, compass/mag and barometer for each EKF lane

rmackay9, 2019-09-02:
We should log which IMU, compass and barometer each EKF lane is using to allow easier debugging of estimation issues.

One way we could do this is to add some extra columns to the NKF4 (1st EKF lane) and NKF9 (2nd EKF lane) messages.  The final columns might be:

- PE : primary EKF lane (renamed current "PI" to "PE")
- UI : used IMU instance
- UM : used compass/mag instance
- UB : used barometer

tridge, 2019-09-02:
yep!

peterbarker, 2019-09-02:
On Sun, 1 Sep 2019, Randy Mackay wrote:

> One way we could do this is to add some extra columns to the NKF4 (1st
> EKF lane) and NKF9 (2nd EKF lane) messages. The final columns might be:

Not to mention NKF14 (3rd EKF lane).

IamPete1, 2021-09-12:
We now have this:
```
// @LoggerMessage: XKFS
// @Description: EKF3 sensor selection
// @Field: TimeUS: Time since system startup
// @Field: C: EKF3 core this data is for
// @Field: MI: compass selection index
// @Field: BI: barometer selection index
// @Field: GI: GPS selection index
// @Field: AI: airspeed selection index
// @Field: SS: Source Set (primary=0/secondary=1/tertiary=2)
```

So the only outstanding thing is IMU, I think that is locked to the core index anyway?

peterbarker, 2021-09-15:
On Sun, 12 Sep 2021, Peter Hall wrote:

> So the only outstanding thing is IMU, I think that is locked to the core index anyway?

Yep.  Give or take a mask, I think.

IamPete1, 2021-10-01:
I think the conclusion is that we now log all of this anyway.

closed 2021-10-01 by IamPete1