ArduPilot/ardupilot #12198: Copter: 3.7/4.0 displays duplicate fence-requires-position pre-arm messages

rmackay9, 2019-09-03:
While testing Copter-3.7-dev in SITL I noticed that these two similar pre-arm messages appear when one would be sufficient.

- "PreArm: fence requires position"
- "PreArm: Fence enabled, need 3D Fix"

peterbarker, 2019-09-03:
On Mon, 2 Sep 2019, Randy Mackay wrote:

>  *  "PreArm: fence requires position"
>  *  "PreArm: Fence enabled, need 3D Fix"

I think one of those indicates we're currently unable to load the polygon 
fence, the other that we need to make sure we're inside it.

rishabsingh3003, 2020-01-11:
I was investigating this issue. I could easily reproduce it in SITL if you try to arm without a healthy 3D fix. 

- `PreArm: fence requires position` is issued from AC_Fence.cpp [here](https://github.com/ArduPilot/ardupilot/blob/cefd991fb0b7a295a4c2c572d4ee4cb309b36381/libraries/AC_Fence/AC_Fence.cpp#L165)

- `PreArm: Fence enabled, need 3D Fix` is issued from AP_Arming.cpp [here](https://github.com/ArduPilot/ardupilot/blob/cefd991fb0b7a295a4c2c572d4ee4cb309b36381/ArduCopter/AP_Arming.cpp#L528)

Since both of them practically check for a valid position estimate from GPS can't we just remove one of the messages altogether? I think we can easily remove the message from AP_Arming.cpp.
I'd like to submit a PR if someone can confirm this.

IamPete1, 2021-08-15:
`PreArm: Fence enabled, need 3D Fix` has gone from the code. I have failed to find the exact PR, but it not there anymore.

closed 2021-08-15 by IamPete1