ArduPilot/ardupilot #12209: Rangefinder in built-in SITL always sends (-1 m) altitude.

MikeCharikov, 2019-09-04:
"range" field is not updated in built-in SITL (SIM_Aircraft.cpp). That's why virtual rangefinder value is always equal to -1.

The idea is to add range calculation to SIM_Aircraft.cpp as 

range = -position.z;

or, for better testing:

range = -position.z - sin((position.x + position.y) / 10) * SCALE_PARAM;

where SCALE_PARAM - some new SIM parameter. 

**Version**
3.6\3.7

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
QUAD(SITL)

**Hardware type**
SITL

khancyr, 2019-09-04:
hello,

Did you enable rangefinder in SITL ? we got autotest for rangefinder on SITL and it is working

MikeCharikov, 2019-09-04:
Hi!

Yes, i did. With the exception that there is no parameter group RNGFND1*, only RNGFND* and RNGFND2*. 

Rangefinder was turned on (corresponding telemetry messages was running) but value was always (-1.0) so rangefinder was in unhealthy state.

When i dig into the code i didn't find where build-in SITL updated "range" field so this line of code in SITL_State.cpp:

_update_rangefinder(_sitl->state.range); 

was always updated with (-1.0)

Can i take a look at the rangefinder autotest?

khancyr, 2019-09-04:
_sitl->state.range is only used if you different from -1 == you are updating is with another simulator see : 
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_SITL/sitl_rangefinder.cpp#L28
So you probably have an issue on your parameters. Did you reboot after applying rangefinder parameters ?

peterbarker, 2019-09-04:
On Wed, 4 Sep 2019, MikeCharikov wrote:

> Can i take a look at the rangefinder autotest?

https://github.com/ardupilot/ardupilot/blob/master/Tools/autotest/arduplane.py#L1244

MikeCharikov, 2019-09-05:
@khancyr 

Ok, i dug it. (Thank's for pointing me at the code line, i overlooked it somehow :) ). Problem still remains but it can be solved by additional parameters tuning (not sure it is the best solution though).

I need to say that i made all my tests with TERRAIN disabled. (i.e. parameter TERRAIN_ENABLE=0).
In that case when UAV is standing on the ground, height_agl can be less than zero (-0.00027), so rangefinder voltage and pin are equal to zero and rangefinder distance_cm is equal to zero too. 

But by default RNGFND_MIN_CM = 20 cm so rangefinder comes to unhealthy state (RangeFinder::RangeFinder_OutOfRangeLow) when UAV is on the ground.

And when i switch to AUTO mode UAV comes to failsafe because AC_WPNav::get_terrain_offset returns false (_rangefinder_healthy == false). 

So, setting RNGFND_MIN_CM = 0 fix this problem but it is pretty unclear. May be we should always send some positive PIN value from sitl_rangefinder?

MikeCharikov, 2019-09-05:
@peterbarker 

Yes, exactly what i was talking about. But this parameter wasn't mentioned in docs (http://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html#adding-a-virtual-rangefinder)

    def set_analog_rangefinder_parameters(self):
        self.set_parameter("RNGFND1_TYPE", 1)
        **self.set_parameter("RNGFND1_MIN_CM", 0)**
        self.set_parameter("RNGFND1_MAX_CM", 4000)
        self.set_parameter("RNGFND1_SCALING", 12.12)
        self.set_parameter("RNGFND1_PIN", 0)

peterbarker, 2019-09-05:
On Thu, 5 Sep 2019, MikeCharikov wrote:

> Yes, exactly what i was talking about. But this parameter wasn't mentioned
> in docs(http://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html#adding
> -a-virtual-rangefinder)
> 
> def set_analog_rangefinder_parameters(self):
>     self.set_parameter("RNGFND1_TYPE", 1)
>     **self.set_parameter("RNGFND1_MIN_CM", 0)**
>     self.set_parameter("RNGFND1_MAX_CM", 4000)
>     self.set_parameter("RNGFND1_SCALING", 12.12)
>     self.set_parameter("RNGFND1_PIN", 0)

Ah.  So does the rangefinder start reading sensible things when you're 
above that altitude?

PRs welcome for the documentation - you can edit it on github.  Would you 
like to do that?

MikeCharikov, 2019-09-05:
Yes, i'll do.

MikeCharikov, 2019-09-05:
>So does the rangefinder start reading sensible things when you're above that altitude?

Rangefinder always reads things, but when it under minimum given altitude - it comes to unhealthy status what causes failsafe.

MikeCharikov, 2019-10-14:
https://github.com/ArduPilot/ardupilot_wiki/pull/2086

closed 2019-10-14 by MikeCharikov