ArduPilot/ardupilot #12222: VTOL Plane: FailSafe during DO_LAND_START  (RTL AUTOLAND - 2)

slmz1, 2019-09-05:
### Bug report

**Issue details**
Issue with VTOL plane when it is finishing AUTO mission (using RTL AUTOLAND - 2) and doing final command VTOL_LAND in QUAD mode, at this point after long flight time Battery FailSafe can be trigged and plane starts transition to fixed wing mode to complete again DO_LAND_START items and finaly landed. It can be dangereous, because the battery is nearly empty.
The same if you change mode to RTL during execution of DO_LAND_START block, the plane return to the beginning of DO_LAND_START.
I think should be - if DO_LAND_START is starting then ignore any failsafe or RTL command.
As normal RTL do (RTL AUTOLAND - 0).

**Version**
Plane 3.9.8

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
flying wing VTOL

**Hardware type**
Pixhawk 2.1 Cube black

pompecukor, 2019-09-08:
@slmz1 

Issue for this has already be open for quite a while. Andrew promised to fix it when he finds the time.

https://github.com/ArduPilot/ardupilot/issues/11090

Naterater, 2019-09-20:
And https://github.com/ArduPilot/ardupilot/issues/10320 from a long while ago.

IamPete1, 2021-07-12:
Should have been fixed by, https://github.com/ArduPilot/ardupilot/pull/13065, you may also have to set `MIS_OPTIONS` = 2 to do distance based calc.

closed 2021-08-15 by IamPete1